opentx/opentx #7717: Name of model not always showing when selecting model on Jumper T16

Gaylian, 2020-06-11:


kilrah, 2020-06-11:
Screenshot? Example? OTX file?

Gaylian, 2020-06-11:
Andre

Screens shots of missing ( most Times) names.

 

Adrian

3djc, 2020-06-11:
No screenshot was supplied,  but are you using supplied sd card ?

kilrah, 2020-06-11:
You did not say what firmware you are running. Try to reflash OpenTX 2.3.7.

closed 2020-06-11 by kilrah

Gaylian, 2020-06-11:
I did send the screen shots but they didn’t appear on web site. I have add them via the web page.

I’m using the files for the SD card that are on the openTX web site. Version 2.3V0029.

I’m using OpenTX 2.3.8 (nightlies). This problem was also on version 2.3.7

Gaylian, 2020-06-11:
Why is it closed. This looks a lot like a opentx problem. Flashing 2.3.7 does not fix the problem.

 

Closed #7717 <https://github.com/opentx/opentx/issues/7717> .

3djc, 2020-06-11:
It is a script that does that, can't remember if it is iNav or Yaapu, but it is not an OpenTX issue