opentx/opentx #7718: Horus X10 Curve messed up

widl84, 2020-06-11:
Hello, my name is Michael and I have a problem with my new Horus X10S Express.

Firmware 2.3.7 is installed.
Companion is 2.3.7
On Windows 10 Home

I want to balance my elevator's servos, but my curve`s are messed up. if one is changed, the other also changes.
Even if the Horus is switched on again, something changes on the curves.

I have read the entry # 6853 but can not confirm that it does not come with slow rotation.
I also made a video of it.

kilrah, 2020-06-11:
Post the video then?

bsongis, 2020-06-11:
Would you try the nightlies, just to check that the issue is gone? Then you return to official 2.3.7, datamodel is compatible anyhow. We are near the 2.3.8 release anyhow.

widl84, 2020-06-12:
[Untitled.zip](https://github.com/opentx/opentx/files/4769603/Untitled.zip)

widl84, 2020-06-12:
Hello I can not generate the error in 2.3.8N again.
can I go flying with the "N" version?
the last version was 2.2.4 but Access doesn't work there and my receivers are already flashed.

3djc, 2020-06-12:
Wait a couple of hours ...

closed 2020-06-12 by 3djc