opentx/opentx #7720: getRSSI() does not go to Zero in companion

TrueBuild, 2020-06-12:
**Describe the bug**
A clear and concise description of what the bug is.
In telemetry lua code getRSSI() does not go to Zero in companion but the old method getValue("RSSI") does go to Zero.
Code snippet provided.
**To Reproduce**
Steps to reproduce the behavior:
Run the the code provided and click the page button to get to the Telemetry screen.
In the Telemetry Simulator reduce the RSSI db selector down towards Zero or just set RSSI to Zero.
getRSSI() will not read/show Zero (0) but getValue("RSSI")  will read/show Zero.

**Expected behavior**
getRSSI() should show the same value as getValue("RSSI")

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**General information**
Sample code works fine on a real X9D
 - OpenTX version 2.3.7
 - Hardware Frsky X9D 
 - OpenTX Companion 2.3.7 on Windows 10

Sample code.

[rssiTs.lua.txt](https://github.com/opentx/opentx/files/4769246/rssiTs.lua.txt)

TrueBuild, 2020-12-10:
Has no one else noticed this?

TrueBuild, 2021-02-15:
This is not a big problem but it make the system inconsistent.

Storchy1, 2021-02-15:
Yes i have noticed that behavior, i made my own getRSSI() function, so i can return 0 if Lua script runs in simulator

TrueBuild, 2021-02-15:
Having no knowledge of how complicated the function getRSSI() is I cannot judge how difficult it would be to correct it.
It just seems a shame when someone has gone to the trouble to create something for us to use we cannot then rely on it ;(

PR #8297 (2021-03-10): Send 0 with LUA getRSSI() when no telemetry is streaming
This fixes #7720
closed 2021-03-10 by bsongis