opentx/opentx #7721: Assigning a receiver number to External Radio System.

Gary-Manuel, 2020-06-12:
I have just tried adding a new model (on Companion) and if I set the **Internal** Radio System to D16 or LR12 protocols, it assigns the next available receiver number. If however, I set the **External** Radio System (with internal OFF) to any of the available protocols, then the receiver number 0 is assigned i.e. no model match. The receiver number needs to be assigned manually in order to make model match work.

I am wondering whether this is an oversight by the OpenTX developers. How difficult would it be to automatically assign a receiver number to the external module, in the same way that they are assigned to the internal module?

One thing that makes me think that this might just be an oversight is that the number assigned to the internal system follows on from the previous one assigned to the external system. They both therefore must be sharing the same numbering sequence.

kilrah, 2020-06-12:
On external there are so many available protocols that work differently that it makes no sense preassigning something. 
And when you switch module type the settings are cleared, which includes the receiver number.

3djc, 2020-06-12:
And you can have several module of the same type you swap too

closed 2021-03-20 by stale[bot]