opentx/opentx #7722: [Companion 2.3 Simu] Trim value spontaneously resets when switching between flight modes

RC-SOAR, 2020-06-12:
**Problem**
Simple setup with two flight modes:
- FM0 has TrmT as 'disabled'
- FM1 has TrmT as 'own trim', value=+100. 

When switching between the two flight modes, the 'own trim' value spontaneously resets to zero, sometimes immediately, other times after a couple of cycles.

So far I can only reproduce this in Companion.

**To Reproduce**
1. Start Companion 2.3.7, with X9D profile
2. Open [this .OTX file](https://www.dropbox.com/s/vy02biivyakev9o/TrimBug_For_OTX_237_Simu.otx?dl=0)
3. Open the simulator
4. Keep an eye on the TrmT graphic (or CH1, which uses TrmT as source)
5. Switch rapidly between between flight modes using SH.

**Expected**: TrmT should alternate between +100 and zero
**Actual**: after a couple of cycles, TrmT shows zero for both FMs.

**General information**
 - Companion 2.3.7 (but also occurs in 2.2.x)
 - Frsky X9D profile
 - Windows 10 X64

bsongis, 2020-12-09:
@dgatf @elecpower would you have time to fix this one?

dgatf, 2020-12-20:
Sure

PR #8172 (2021-02-02): Remove retry timer for set trim on simulator
Fixes #7722

The issue only affects the simulator, not the firmware

The delay timer applied to the trim is too long, so it may be applied to the wrong flight mode when switching fast between flight modes in the simulator

BTW... why is needed a delay for the trim on the simulator?
closed 2021-02-02 by bsongis