opentx/opentx #7730: TX16S new firmware 2.3.8 problem

E120103016, 2020-06-13:
Hello there:
There is a problem with TX16S new firmware 2.3.8 today. Download version: otx (0541177). Variant 219 Date: 2020.06.13:19:02
After burning the remote control, it is incompatible with the current firmware (bug). There is no way to try. Please tell me how to deal with thank you!

3djc, 2020-06-13:
Please wait the firmware server are back again, we have disabled them for now

And also, please describe your problem. What you are trying to do, what result you expect, what do you get. Things like "it doesn't work" or "it is incompatible" do NOT help understanding your issue

E120103016, 2020-06-13:
I am afraid that the future 2.4.0 firmware will also be incompatible.

3djc, 2020-06-13:
2.3.8 is compatible with TX16S, so will 2.4.x, so will 2.5.x

E120103016, 2020-06-13:
Thanks for your notice

E120103016, 2020-06-13:
The problem has been solved thanks!

closed 2020-06-13 by E120103016