opentx/opentx #7731: Horus X10S 2.3.8 firmware not compatible

robertw2a, 2020-06-13:
Trying to update Horus X10S with 2.3.8 from 2.3.7 OpenTX Companion says file not compatible with existing firmware. Downloaded with R9flex and lua options
[image]
[image]

3djc, 2020-06-13:
Yes this is due to a change that was introduced in this very, please uncheck “check hardware compatibility” for this flash

robertw2a, 2020-06-13:
OK will try, just did not want to corrupt or damage the radio

robertw2a, 2020-06-13:
Thanks, that worked afterwards it will  flash with "check hardware compatibility” checked

commit 23aaafbf (2020-06-14): Fixes #7731
closed 2020-06-14 by bsongis