opentx/opentx #7733: [Companion 2.3.8 Simu] Mixer values collapse to zero during FM fade in/out

RC-SOAR, 2020-06-13:
This issue is new in Companion 2.3.8 (no problem in 2.3.7), unfortunately it completely breaks the sim for many of my templates, setting off wild oscillations in multiple channels. So far, I've not been able to reproduce this issue on the radio, but I wanted to report it before doing further testing.

**The problem**
When activating a flight mode with non-zero fade in/out, mixer values collapse to zero during the transition.

**To Reproduce**
Steps to reproduce the behavior:
1. Download [ this file (minimalist example)](https://www.dropbox.com/s/rlygv36hl5dbyxz/FMslow_bug_238_T16.otx?dl=0)
2. Start Companion 2.3.8 with Jumper T16 profile and open the file
3. Pull SH to switch flight modes while monitoring CH1

**Expected behavior**
CH1 mixer value should remain constant, instead it collapses to zero during the fade in/out (set to 0.3 secs in FM1).

**General information**
 - Companion 2.3.8 (release version)
 - Jumper T16 profile

bsongis, 2020-06-14:
Perhaps because we are in 64 bits. We didn't test the change on the simulator, only on the radio!

dgatf, 2020-06-14:
This guarantees a positive bit shift:
`
(17 - __builtin_clz(abs(chans[i])))` --> `(__builtin_clz(abs(chans[i])) >= 17 ? 0 : 17 - __builtin_clz(abs(chans[i])))`

bsongis, 2020-06-14:
critical bug then :(

bsongis, 2020-06-14:
we have to release 2.3.9 today, no choice

closed 2020-06-14 by bsongis