opentx/opentx #7735: [Companion 2.3.8] Cut on SF prompts Clear dialog

Mateyhv, 2020-06-14:
Small bug on Companion 2.3.8. Cut works as expected but prompts the dialog related to Clearing a SF. Tested on LS and its OK there.
[image]
[image]
Thanks

RC-SOAR, 2020-06-14:
I've also encountered an issue in Companion 2.3.8, where Clear on a single LS causes multiple (but not all) LS's to be cleared.  Could this be related?

elecpower, 2020-06-14:
Cut performs a copy and clear and thus the message from the the underlying clear function. Probably a better message required.

Mateyhv, 2020-06-14:
@elecpower previous Companion releases prompt no message for Cut which I believe is the correct way since its not a critical action.

Clear on the other hand cannot be undone so the warning message is welcome.

elecpower, 2020-06-14:
@Mateyhv as of 2.3.8 the new menus now chase all references to the edited entry (except Telemetry Sensors in a future version) and update them in an effort to minimise breakages and thus manual updates.

elecpower, 2020-06-14:
@Mateyhv and so cut can not be easily undone if the cut entry is referenced elsewhere

Mateyhv, 2020-06-14:
@elecpower thanks for the explanation. Closing the issue!

closed 2020-06-14 by Mateyhv