opentx/opentx #7737: Companion : issue in SF set failsafe

3djc, 2020-06-14:
Companion does offer "set failsafe internal" and "set failsafe external" when only one should be proposed

[image]

Exists in 2.3.9 and has probably be present for some time

elecpower, 2020-09-17:
So does Companion simulator and X9D+ radio even if an internal and/or external module is no configured.

kilrah, 2020-09-17:
Thing is we used to have separate failsafes for internal and external and it's not the case anymore.

bsongis, 2020-12-09:
@elecpower it should be possible to configure the failsafe even if there is no module configured

bsongis, 2020-12-09:
@dgatf @elecpower this one for you? let me know!

elecpower, 2020-12-09:
@bsongis I'll take it

PR #8156 (2020-12-18): Remove module choice from special function set failsafe
Fixes #7737
closed 2020-12-18 by bsongis