opentx/opentx #7738: Model Wizard does not work in Horus X12S

Luznatural, 2020-06-14:
Description:
With Open TX version 2.3.9, when you try to create a new model in Horus X12S, the screen where the model wizard should allow you to select plane, glider, or drone, appears just empty. So, you cannot make the wizzard

**To Reproduce**
Steps to reproduce the behavior:
1. select MDL. when the list of models appear, 
2. press and hold the enter wheel
3. Select create a new model
4. the screen goes empty with no option to select

**General information**
 - OpenTX version 2.3.9
 - Hardware Frsky X12S
 - OpenTX Companion 2.3.9 Windows

3djc, 2020-06-14:
You have not installed the sdcard pack correctly I’m afraid

closed 2020-06-14 by 3djc

Luznatural, 2020-06-14:
I did check twice. I have in the script folder the wizard folder, and in the folder I have both folders, glider and plane

Luznatural, 2020-06-14:
both folders appear to have the correct files on them

Luznatural, 2020-06-14:
I checked on my second radio, a Frsky X9D+, and works correctly