opentx/opentx #7739: "Path too long" error when calling playFile() with 37+ characters

paulo-sousa, 2020-06-15:
I'm using **iNav with Lua Telemetry Script** and I upgraded from OpenTX 2.3.7 to OpenTX 2.3.8 on my Taranis Q X7 just after 2.3.8 was released. The telemetry scripts were working fine in OpenTX 2.3.7, but stopped working after the upgrade. I retested with OpenTX 2.3.9 and it still fails.

The telemetry script is calling playFile() with the following parameter **playFile("/SCRIPTS/TELEMETRY/iNav/en/engarm.wav")**, and it fails with the error **"Path too long"**.

**To Reproduce**
Steps to reproduce the behavior, as end-user of iNav Telemetry:
1. Upgrade a Taranis Q X7 to OpenTX 2.3.8 or OpenTX 2.3.9.
2. Copy the iNav Telemetry Script 1.7.2 (LuaTelementry-en.zip) over to SD Card.
3. Configure a model with FrSky Telemetry.
4. Activate iNav Telemetry Script.
5. Turn model on, so that telemetry is received.
6. Perform a function that would trigger a voice alert (e.g. "Arm Motors").
7. "Path too long" error message is displayed on screen.

Alternatively, create a simple Lua test script and place it in the "SCRIPTS/TELEMETRY" folder:
```
-- Simple playFile Test

local function background()

end

local function run(event)
--      Files with < 37 characters work in OpenTX 2.3.9
--	playFile("/SCRIPTS/TELEMETRY/iNav/en/good.wav")
--	playFile("/SCRIPTS/TELEMETRY/iNav/en/fsoff.wav")

--      Files with >= 37 characters fail in OpenTX 2.3.9
--	playFile("/SCRIPTS/TELEMETRY/iNav/en/engarm.wav")
	playFile("/SCRIPTS/TELEMETRY/iNav/en/hedhlda.wav")
	return 0
end

return { run = run, background = background }
```

**Expected behavior**
The sound alert should be heard and no error message should be shown.

**Screenshots**
[image]

**General information**
 - OpenTX version 2.3.8 & OpenTX version 2.3.9
 - Hardware FrSky Taranis Q X7
 - OpenTX Companion 2.3.8

3djc, 2020-06-15:
Indeed, this is an undesired impact of fixing an issue for Horus that affect 128x64 radios

closed 2020-06-15 by bsongis

yesthisusernameistaken, 2020-07-25:
Having this issue on my T12 with OpenTX 2.3.9. 
Am I to understand that with 2.3.10 will have a fix for this?

3djc, 2020-07-25:
Yes

Black-Eagle954, 2020-09-01:
Having the same "Path Too Long" error message problem in the q x7 Tx with Open Tx 2.3.9, Inav 2.5.2 and Inav Lua Telemetry 1.7.2 once any switch is turned on like ARM.

commit 4cca618f (2020-06-15): Fixes #7739