opentx/opentx #7742: Updating firmware in the OpenTX boot loader mode using Companion "bricks" taranis.

L2-Max, 2020-06-15:
Yesterday, I tried to update my radio using Companion's "Write Firmware" feature several times but no success. The Companion is always finish updating with success however, when I disconnect USB cable and press "Exit" menu on the bootloader screen my radio just turn off the screen and nothing happen. The power button remains lit blue. Long press or short press on the power button does not do anything and the only way to turn it off is to disconnect it from battery power. If I connect battery again and press power button the radio just clicks and nothing happens. The only way to recover radio after this update is to enter The Open TX Bootloader mode again and to flash exactly the same firmware directly from SD card.

So, I would expect updating firmware using Companion's feature works well as expected and not to "brick" the radio.

 - OpenTX version 2.3.7/8/9
 - Taranis X9D+ SE 2019
 - OpenTX Companion 2.3.7/8/9

kilrah, 2020-06-15:
What PC OS?

Do you correctly "eject" the disk before disconnecting USB?

And yes this method is known not to be very reliable, if it doesn't work use one of the other 2 available ones.

L2-Max, 2020-06-15:
My OS is Windows 10. I did both, correct eject as well just simply pull USB cable out of my radio. Every time result was the same. Yesterday, I had a time to play with that 15 - 20 times and it did not flash firmware correctly even once. Every time I ended up with "bricked" radio.

So, then the question is: Why this not very reliable firmware update method still exists?

kilrah, 2020-06-15:
Problems are normally very intermittent in normal cases, for some people it maybe fails 1 time out of 10, for most it always works. 

Maybe you have some antivirus or such that intercepts the process and causes trouble...

L2-Max, 2020-06-15:
Understand you.

I have just gave it yet another try and looks like I got why I had so many unsuccessful firmware updates. It looks like either Companion or Taranis does not actually overwrite (using update Method 2) firmware if it is not changed (despite the fact it is corrupted on device) and if previous firmware update failed then any consecutive attempts to flash the same firmware do nothing.

What I did just now was (using Method 2):
- Flashed 2.3.7, eject ; success
- Flashed 2.3.9, eject ; success
- Flashed 2.3.7, eject ; success
- Flashed 2.3.9, did not eject ; fail
- Flashed 2.3.9, eject ; fail
- Flashed 2.3.9, eject ; fail
... the same a few more times ...
- Flashed 2.3.7, eject ; success

If that is true, it would be better if Companion or bootloader, which simulates internal drive, calculate firmware checksum to detect firmware difference rather than simply compare version numbers.

kullfpv, 2020-07-03:
what is method 2? , I am in your same problem, you have managed to put 2.3.9, greetings.

> Understand you.
>
> I have just gave it yet another try and looks like I got why I had so many unsuccessful firmware updates. It looks like either Companion or Taranis does not actually overwrite [...]

3djc, 2020-07-03:
Put firmware file in SDcard FIRMWARE directory, disconnect radio, start in bootloader, followonscreen instruction for Flash firmware

kullfpv, 2020-07-03:
> Put firmware file in SDcard FIRMWARE directory, disconnect radio, start in bootloader, followonscreen instruction for Flash firmware

everything correct the station is 2.3.9, thanks, I tried it by opentx in 2.3.7 and error, opentx should review this bug, regards.

vanpit, 2020-07-14:
Same problem here. Radio Taranis X9D+ SE 2019. Flashing in OpenTX bootloader + USB randomly crashes with symptoms described by L2-Max. Flash by STM32-bootloader or by SD card is always successful. Tested on Companion 2.3.9.

Nightcr4wler, 2020-09-25:
Hi L2-Max

The same was happened with me on Taranis X9D 2019 updated with Companion on Windows 10, after many tries i think that i found a solution.

Taranis X9D 2019 can be flashed using the Companion on Windows 10 only fixing DFU driver on OS. You need to update the STM32 Bootloader driver correctly and the driver it should appear on Windows Device Manager like a Universal Serial Bus device and not Universal Serial Bus controller, the right way to do that is download the file driver on web and restart the Windows in Advanced Mode to Disable Driver Signature Enforcement, has some videos on youtube to do it, after reboot plug the USB on Taranis and the power button will show green light and the Windows will show on Device Manager a new Universal Serial Bus Controller try to update the driver with the file downloaded and now should appear like a Universal Serial Bus device and the Companion works to flash the radio.

3djc, 2020-09-26:
That's a very bad solution, and disabling signature is NOT needed. Put the firmware on the SD card and flash from Radio bootloader

venom-code, 2020-09-26:
1) i also managed to flash 2.3.9 with the sd card method with the companion it kept getting bricked, everything seems okay other than the splash screen because with the sd card method i am just getting the default OPENTX welcome screen regardless how many times i tried, the firmware was downloaded through companion with all my options selected like lua, noheli etc, those seem to be correct but only thing i cant get is my splash screen back, any solutions? and when do you guys think this issue will be resolved and we can use companion to flash again

3djc, 2020-09-26:
Flash from sd it is a much simpler and less error prone method

venom-code, 2020-09-26:
yes thats what i did as well, so how do i get my splash screen through the sd card method if you can elaborate as i already tried many times the fw gets updated fine through sd card but the splash screen is the default opentx one, any way to correct this?

kilrah, 2020-09-26:
Settings -> Edit Radio Splash image in companion can replace the built-in splash in a firmware file.

venom-code, 2020-09-26:
yes in the companion i can see my splash screen which always worked normally when flashing through the companion itself but now that the radio is getting bricked everytime when using the companion to flash fw the only option was to use the sd card method and through this method the splash screen i had chosen does not show this is what i am saying, and if there is a work around it would be great to know

kilrah, 2020-09-26:
Read my message....

venom-code, 2020-09-26:
i have bro, infact i already tried that a couple of times with no success that is why ii asked again :)
i will give it a shot again in a while

3djc, 2020-09-26:
No you haven’t, at least not correctly. He told you you can modify the image on the firmware file before putting it on sd to flash from there

venom-code, 2020-09-26:
sir if you have read my previous msgs i have already mentioned that, when i download the fw through the companion to put on the sd card i chose my
1-Radio type (again to be sure)
2-Menu Language
3-Build Options
4-Splash screen (chose the file i have on my pc which was downloaded from open tx org

after doing the following i press OK and then choose the download button arrow which downloads the fw and i copy it on the sd card in the FIRMWARE folder, eject Taranis properly and while in bootloader i choose flash firmware, use the fw file i copied on to the sd card, the flash is a success it starts fine with all models and configuration intact BUT the splash screen is not which i chose in the companion before downloading the fw, (i hope this is what you guys are suggesting as well and i explain in detail so all steps are known as to what i did) , my apologies to everyone if my many msgs bothered any of you friends

kilrah, 2020-09-26:
- This place is not for user help
- I literally told you what to click in my message already

venom-code, 2020-09-26:
i know, i really appreciate the help, i will try to resolve the issue for myself, i have already done this method of going into settings (not the scroll whee) and click on edit splash screen and i select the one i want and yet it does not work, i hope Opentx and Taranis find a proper solution to this issue, i tried updating the fw on my other tx a qx7 with the same companion (diff radio profiles offcourse) and everything went smoothly on that... hope others can take advantage of the above points you abd 3djc have mentioned and what i tried

kreidler, 2020-12-11:
Just to push the issue up.

- OpenTX version 2.3.10
- Frsky X9D+ SE 2019 / X9D+ / X7

Thank you for the idea with the splash image editor. Works well for me but I do think that it should not be the intension to say: "Please use editor and bootloader flash option" even if the flash procedure does not work with origin splash.
I can reproduce this issue with the X9D+ 2019 edition only. The older X9D+ as well as a X7 do not have any problems flashing with or without splash screens via companion.

richard-scott, 2021-07-17:
I’ve a bricked Radiomaster TX16S after attempting an update with the latest release. I’ve got no boot loader menu using the trim at power on so can’t flash via SD card can Flash via DFU but nothing works. This is both on Linux and Win10 after using zdiag to fix drivers.

I get a white lcd and nothing else.

3djc, 2021-07-17:
Then you are likely not using the right files, Tx16s cannot be bricked.
please paste here your complete dfu log

rotorman, 2021-07-18:
@richard-scott I wrote for EdgeTX Wiki some easy to follow "unbricking" instructions - these apply to OpenTX the same way. Maybe these will help you to sort out the issue you are having.
[https://github.com/EdgeTX/edgetx/wiki/Unbrick-your-radio](https://github.com/EdgeTX/edgetx/wiki/Unbrick-your-radio)

The Device Firmware Upgrade (DFU) is a hardware feature of the STM32 chip (the main microcontroller in TX16S radio) that cannot be altered, erased or otherwise tampered with, thus it is always there and will allow you to easily recover from flashing mishaps.