opentx/opentx #7744: Hott Telemetry Value BCpT fault scaling

Ziege-One, 2020-06-15:
I use the Hott System with the Radiomaster TX16s and an GAM Sensor on the Receiver side. The 
The value used mAh (BCpT) shows 0.01mAh, but in real it is 10mAh.

Here in opentx/radio/src/telemetry/hott.cpp you see the scaling "battery capacity in 10mAh steps"

`          // packet[11] uint8_t batt_cap_L;     //#11 battery capacity in 10mAh steps
          // packet[12] uint8_t batt_cap_H;     //#12
          value = packet[11] + (packet[12] << 8);
          sensor = getHottSensor(HOTT_ID_BCAPA);`

Thank you for great Job on OpenTX

**General information**
 - OpenTX version (2.3.9-otx)
 - Hardware (Radiomaster TX16s)
 - OpenTX Companion (2.3.9 on Windows)

3djc, 2020-06-16:
@Ziege-One I have pushed a fix that will be in next nightly (247), please test and let us know how it worked :)

closed 2020-06-16 by 3djc

Ziege-One, 2020-06-16:
@3djc Thank you
you have forgotten one line at 614.

3 Sensors have the used mAh ESC, GAM and EAM.
I will test it

Ziege-One, 2020-06-17:
Hi,
nightly (247) install and tested. It works very well. I will check the other Values and will report.
Thank you form "Speedy" Update 👍

commit 9c70be75 (2020-06-16): HoTT BCpT scaling fix
This fixes #7744