opentx/opentx #7745: Possible control issue with X10 Express/R9M 2019

kilrah, 2020-06-16:
Posted in Discord

>  2.3.9 keep losing signal on FrSky R9 Stab, also problems on R9 Slim+ OTA and S6R. Went flying and crashing due to issues. Reinstalled 2.3.7 and everything settled down and back to how it was.
>  Was using Horus X10 Express and various receivers, all FrSky, R9 Stab, X8R, R9 Slim+ OTA, all kept losing signals and wanting to reset, drove me nuts! On the R9 Stab, every time I went to try to do calibration, within 30 seconds it had switched off, lost signal, came back on and would not let me complete calibration.  Once I went back to 2.3.7 I was able to calibrate first time, all the other receivers acted normal.
> R9M 2019 ACCESS 2.1.0
>I flew without problems 2 days ago, same models, same conditions. Updated to 2.3.9 and my Tundra(s), Darts, G2 Ranger, GT Talon all lost signals! Explain. Reinstalled 2.3.7 and all now fly as they did before without problems.
> tried different R9 Stab (I have three) on three different models and all three reacted the same so I reinstalled 2.3.7 and rechecked them and all three now work as they did before 2.3.9. I did the same on the S6R and R9 Slim+ OTA.
> The R9 Stab kept stopping and starting, like "lost signal" "regain signal"

Adelaxie, 2020-06-16:
Test with X10 Express opentx 2.3.9 
ISRM  2.1.0 FCC,RX4R 2.1.0 under ACCST
R9M 2019 ACCESS 1.3.0 EU,R9 SX 1.3.0
All works.

bsongis, 2020-06-19:
OP answer today: 

>On my Horus X10 Express I have:  ISRM-S-X10E_2.1.0. On my R9M2019 I have:  R9M2019_LBT. I have ACESS versions on my R9 Stab OTA/R9 Slim+ OTA/R9 Slim+: FW-R9Stab-ACCESS_v1.3.1/FW-R9Slim+OTA-ACCESS_v1.3.1/FW-R9Slim-ACCESS_v1.3.1. On my S6R and S8R I have been running S6R_ACCST_2.1.0_LBT.frk.
>I was running 2.3.9 Companion, see attached, although this is a screenshot of 2.37 these are the settings I used. I have since gone back to 2.3.7 both on Companion and on my Horus and I flew all the same planes at the same flying site and had no issues (well one issue on one plane which was going out of range which was my fault). All the planes electronics behaved as they should have. So I know it is not my transmitter as it works fine of 2.3.7.
>I know I cannot go flying today so I am tempted to try to reinstall 2.3.9 to see if I had installed it correctly and if the problems reappear as I am now confident that if they do reappear I can at least go back to 2.3.7 knowing I can still fly. I hope this helps?
>I only reported the problem initially as I thought I was helping but did feel like I was the bad guy for saying something, which is why I did not reply immediately.

3djc, 2020-07-03:
OP update

> Good morning. Since I have brought 2 of these units (R9 Stab OTA), I decided to redo a "stick" calibration again on the 2nd unit in a Dynam Primo plane and to my surprise it worked and calibrated as intended. (On these units, you have to re-do a stick calibration every time you re-trim a plane or before flying it for the first time, so the TX knows the servo limits and that is the calibration I have been talking about.)
> So my conclusion is that I must have a bad unit

closed 2020-07-03 by kilrah