opentx/opentx #7748: Wrong Italian wav file for km/h and low battery.

fdapport, 2020-06-17:
Hardware: Taranix XD9+
Firmware: 2.3.9
SD Card Files: sdcard-212x64-2.3V0029.zip
Under: SOUNDS/IT/SYSTEM
Files: kph0.wav,  kph1.wav and lowbatt.wav are wrong pronunciation.

The correct one are stored in SD Card files: sdcard-212x64-2.3V0026.zip
Under: \SOUNDS\it\SYSTEM

We could have other files with wrong Italian pronunciation, I didn’t check all.
Franco

3djc, 2020-06-17:
The only difference between 26 and 29 as far as Italian is concerned are here : https://github.com/opentx/opentx/commit/f0a1c65d952a7d7df32ca075dfba8d0251163a1d#diff-11b76c79cb42fd865dcb3825dc1c3c72

The one you mentioned haven’t been changed apparently, so it might well be some changes in the online generation tool we use

fdapport, 2020-06-17:
On the radio and as well on the PC the wav file say "take kilometri orari" instead of "kilometri orari".
You can try by yourself, you could find that are different.

Also the files have different proprieties:

[image]
[image]

Thanks,
Franco

closed 2021-03-20 by stale[bot]