opentx/opentx #7750: OpenTx Companion can crash your radio even when flashing the right firmware.

KavinUser, 2020-06-18:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
In my case, flashing 2.3.9 from the Companion on to my Taranis X9D+ 2019 bricked my radio.   In talking to the developers, this issue does not appear to be limited to that radio.

A concise warning/note of how to fix that result if it happens to the user in flashing over Companion could be very helpful and time saving.

Toogo, 2020-06-20:
Same radio, same problem!

3djc, 2020-06-20:
Copy firmware to FIRMWARE dir on your SD and flash using bootloader 'Flash firmware' menu

RipperDrone, 2020-06-22:
flashed 2.3.9 on X9D+ (2013), and it worked nicely. Used bootloader mode to do so...

3djc, 2020-10-03:
Duplicate #7742

closed 2020-10-03 by 3djc