opentx/opentx #7751: TX16S UART power issue

xkriton, 2020-06-18:
Radiomaster TX16S OpenTX 2.3.8 stable (noheli, lua, flexr9m)
Both bottom UART outputs lose power after selecting Telemetry Mirror option.

Behavior when selecting options:
OFF (no power) -> Telem Mirror (no) -> Telemetry In (no) -> SBUS Trainer (no) -> LUA (powered!)
when going back:
LUA (powered) -> SBUS (powered) ->Tele In (powered) -> Telem Mirror (lost power) -> OFF (no power)

3djc, 2020-06-18:
ok, great time to review which of those more should have power. Should they all be powered but OFF ?

xkriton, 2020-06-18:
Well, hard to tell. Just finished testing with stock FW from Radiomaster and there it keeps power even with "OFF" selected.
In my case I am focusing on "taranis style" BT Telemetry Mirror, which requires power for hooked up BT module. Honestly I did not even investigated possibilities of Tele In/SBUS/LUA options yet :-)

xkriton, 2020-06-22:
After quick test with custom compiled FW (AUX_SERIAL_POWER_ON) there is everything OK with power for UARTs and I have telemetry mirror working over attached BT module.

closed 2020-06-25 by 3djc

skaman82, 2020-06-27:
@xkriton any chance you can send me this firmware where the uart is powered on telemetry mirror (using crossfire)?

ideally with this options selected:
[image]

3djc, 2020-06-27:
Nightly

skaman82, 2020-06-27:
@3djc thanks - how does this work exactly? - I can only enable release candidates in companion (there are none)

3djc, 2020-06-27:
You need nightly companion : https://downloads.open-tx.org/2.3/nightlies/companion/windows/
Set to nightly after install

skaman82, 2020-06-29:
thank you very much

al21883, 2020-09-20:
Radiomaster TX16S OpenTX 2.3.9

On all UART options except OFF I have the left UART powered and the right UART without power. It's supposed to work that way?

Thanks!

commit ed1c1f06 (2020-06-25): Fix AUX/AUX2 power handling.
This fixes #7751