opentx/opentx #7752: Telemetry Logging - change requested

mpjf01, 2020-06-18:
In testing different combinations of receiver and sensors with the same model I note that the log file does not reflect any changes that are made to telemetry sensors - new or removed - until the next day (I think).

Presumably the format of the log file is fixed by what sensors are there when the file is created at the start of each day's flying with that model. 

Would it be possible to create a new log file whenever a sensor discovery is done, or at the start of each days session if no discoveries are done?

That way the log file would always show the logging data that the user had asked for.

or -- is there some alternative mechanism other than having to rename log files multiple times during a test session?

lshems, 2020-09-17:
Just remove or rename the logfile from the sdcard file browser

mpjf01, 2020-09-18:
Please read my request, particularly the last sentence.

aeronaut999, 2020-09-21:
This a great suggestion; I guess I don't check log files enough to realize that this was going on.

Miami-Mike, 2020-09-22:
How about a special function called "Close Telem" for "close telemetry file"? Maybe it could close and rename the current file from "ModelName-2020-09-22.csv" to ""ModelName-2020-09-22-1.csv", where the last number would be the lowest available number for that filename in the LOGS folder.

mpjf01, 2021-03-29:
#8345 might be related

MarkusPfaff, 2021-08-13:
Whenever you change the name of a logged sensor or you add a new sensor or you delete an existing sensor the currently used logfile should be closed automatically and a new one should be opened. The files generated could be named by -1, -2, -3 as proposed above already.

Currently the log file will get corrupted under the situations described above.

MarkusPfaff, 2022-04-17:
The issue still persists.