opentx/opentx #7755: EU option removed from X9D+ 2019 version 2.3.6

barryv54, 2020-06-19:
Hi Guys, Version 2.3.6 of OPenTx removed the EU option and as a consequence we are no longer  is able to deselect the option to enable D8 mode. The 'core' software is still EU/LBT when this is selected so we are not contravening any laws. D8 is still required to enable users to connect to the older D8 receivers (that are still available) and also fly someof the quads on the market that have inbuilt receivers.

Can we have this reinstated. Using the method of deselecting EU when creating firmware was an accepted method of enabling D8 mode without making the TX illegal.

3djc, 2020-06-19:
X9D 2019 cannot do D8, since ISRM do not support it. EU doesn't therefore make any sense

closed 2020-06-19 by 3djc

barryv54, 2020-06-19:
Prior to upgrading to 2.3.6 I was able to use D8 mode by disabling the EU option. This process is highlighted on Youtube and a number of people have followed this 'Hack' so dont understand your comment, have you tried it?
Thx

3djc, 2020-06-19:
This does NOT apply to 2019 version of x9d on which FrSky has removed support for D8

barryv54, 2020-06-19:
Thx 3djc, my apologies the radio I used was a X9D+, changed to the 2019 se and now lost functionality. Does that mean that the D8 mode is no longer supported on both the EU and FCC 2019 Transmitters. If thats the case I can see there being a lot of unhappy FRSKYY users. Once again thanks for your help and my apologies

3djc, 2020-06-19:
Yes, D8 is NOT supported at all on 2019 version, FCC or EU doesn't matter

pMatviienko, 2021-03-16:
Hi everyone, D8 not working anymore understood, but what about D16 mode? My 2019 x9d had it initially but after firware update there are no d16 mode anymore. is tere some way to bring it back?

kilrah, 2021-03-16:
It's never been removed.

shredman01, 2023-01-10:
I also bought a Taranis x9D+ 2019 and I'm annoyed because I didn't know that the D8 mode was no longer available in the 2019 models. If I had known that before, then I would have bought the RadioMaster TX16S Mark II. I think removing the D8 protocol from the hardware because it is supposedly illegal in the EU is a cheap excuse from FRSKY. Here you simply want to force the customer to buy completely new and much more expensive ACCST+ Access receivers. It would certainly not have been a problem to unlock the D8 mode with LBT in the new ISRM transmitter module but no, here the customer is clearly patronized in order to cash in on him. Older transmitters that bought before 01.01.2015 with XJT Modules are supporting D16, LR12 and D8 Mode with LBT and are not illegal in EU when you use it. Why not in newer 2019 Transmitters with ISRM modules ? ;-)