opentx/opentx #7756: radiomasters TX16S multi protocol module

Dennis38, 2020-06-19:
possible issue can't change the protocol in the multi module it only has assan as selectable nothing else

3djc, 2020-06-20:
Please provide more details, like exact radio firmware version, module version

Dennis38, 2020-06-20:
opentx-tx 16s-radiomaster (7ee5b1d0
2020-04-28
uid 002c0046 30345105 36323435
opts: crossfire,internalmulti,multimodule,luac
Thats everything in the information section of the radio 
Trying to figure out if this is a firmware problem or hardware problem hope you can help 
thanks for looking at the issue

3djc, 2020-06-20:
This is ok, what version does the module show . (In model set, where you set the protocol)

Dennis38, 2020-06-20:
So in the protocol multi module screen there is no version( only this statement assan ) 
and next line states invalid protocol.
 I took out the SD card and opened the firmware folder inside i found the following 
firmware_TX_16S_1_EN_2048 this listed twice
next line 
Multi-STM-openTX-aetr-inv-13.0.91
multi-STM-openTX-aetr-noinv-13.0.91
I hope this is what you are asking for 
Thanks again for looking at it

3djc, 2020-06-20:
It seems quite clear there is an issue with your radio. Please contact your sellers for warranty

closed 2020-06-20 by 3djc

Dennis38, 2020-06-20:
Okay so it is a hardware issue i will go ahead and contact the seller i just wanted to be sure of what's going on before i made a claim many thanks for the help.