opentx/opentx #7757: New telemetry screen layout for 480x272

KissmanCZ, 2020-06-20:
Hello, I would like to propose a new layout for telemetry screen. My idea is 4+2 or 2+4 layout. It is difficult to add all wanted values to one screen especially when some telemetry sensors cant be used with small cell (GPS for example). 2 pages is not always way to go.

[image]

robustini, 2020-06-20:
I agree, good idea.

KissmanCZ, 2020-06-20:
I am just thinking... what about this way:

[image]

So two new layouts (4+2 & 4+3) and **add option** to rotate layout horizontaly or vertically? Just two checkboxes below... So user will be able to place from 4 different styles for 2 widgets. Can work also for other layouts.

robustini, 2020-06-22:
It would be very useful, I hope it will be taken into consideration.

mhfalken, 2020-07-11:
Yes, more layouts would be nice, but also more layouts without the trim/sliders as they take up a lot of real estate and are never used on drones. I know they can be disabled, but that doesn't give more space to the widget, so extra widgets with larger boxes, so more of the screen can be used.

Hobby4life, 2020-09-24:
I have not seen this request but similar to mine: https://github.com/opentx/opentx/issues/7921

Pleas devs add more type of layouts :-)

Hobby4life, 2020-10-14:
I have created my own layouts:

https://github.com/Hobby4life/opentx/tree/2.3/radio/src/gui/480x272/layouts

https://www.youtube.com/watch?v=NJXQOz2F65A

PR #7992 (2020-10-16): Add 4+2 super flexi layout
This add a configurable 4+2 widget, that adapts to use the available space based on its option.

This closes #7757

Default option
[image]

Mirror
[image]

Without trims
[image]

Without trims and sliders
[image]

Lets go crazy !
[image]
closed 2020-10-16 by bsongis