opentx/opentx #7763: Download older firmware using Companion

habbiebabbie, 2020-06-23:
Hi !

I recently thought I had a problem with 2.3.9 and wanted to download the 2.3.7 firmware version using Companion. I did not find an option to specify which firmware version I'd like to download. It always points to the most recent version.
Am I missing something ?

Regards,
René

3djc, 2020-06-23:
It is not possible to download an old version by design. Firmware are compiled on demand, and you always get the lastest one in the branch (2.3.x, 2.2.x)

closed 2020-06-23 by 3djc

Hobbypilots, 2021-08-13:
> It is not possible to download an old version by design. Firmware are compiled on demand, and you always get the lastest one in the branch (2.3.x, 2.2.x)
agree that opentx down does not allow users to download the previous version, but is there any way of doing it.

bigtwin, 2021-08-13:
You can consult your favorit support forum and simply ask if somebody still has your desired version with your wanted features.

gratefulfrog, 2022-08-18:
Is this still the case?  The latest versions of opentTX do not work properly with the multimodule, the last working version is 1.3.13. I would really like to be able to configure it myself and not depend on other's versions..

3djc, 2022-08-18:
So untrue,what issue are there with multi on latest version ? Most of the supported radio are now actually multi based ...