opentx/opentx #7764: Jumper T16, can't assign certain PNG images to models

fiechr, 2020-06-23:
I'm running OpenTX 2.3.9 on my Jumper T16 and I had a similar issue before, but it somehow did go away by itself, this time it doesn't...

I've downloaded all these pictures for my models from https://www.air-rc.com/

The problem basically is, that the images don't show up in the drop down menu where I can select the image by name. This doesn't work on the transmitter or the companion software. I've restarted the transmitter and copied the files again, but that didn't change anything.

I can see the images though, on the file system view of the global settings menu. But if I open the context menu, I don't get the "Assign bitmap" function, while on other, working images, this menu entry shows up.

All the PNG images have the same resolution and have also similar file sizes.

I've attached four images, two that work fine and two that don't:

These two do work fine and are assigned to a model each:
[image]
[image]

These two are the ones I can't see and/or assign:
[image]
[image]

3djc, 2020-06-23:
Name too long ?

3djc, 2020-06-23:
Tested here ok when named t16.png

closed 2020-06-23 by 3djc

fiechr, 2020-06-23:
Yeah, indeed, name was more than 8(?) characters.

To be honest, this is pretty confusing, since there is no warning or information (like in the readme.txt in the folder), that there is a filename length limit. Especially since I can copy files with long names over without errors and even look at them (in case of images) in the global file menu. I'm sure I hat this issue before, but didn't realized it then.