opentx/opentx #7766: Internal Bluetooth disconnecting after several seconds

skaman82, 2020-06-24:
**Describe the bug**
I am using the para BT module in my X-Lite and Full CRSF TX. I set BT to "telemetry" and I can connect to it normally using Android Telemetry Viewer App or connect to it with an HM-10. BT is paired and pin is also entered. At first I am getting data normally but after several seconds (sometimes 5 - sometimes 60) the connection is lost. 

The Android App must be reconnected manually until it loses connection again after max 1minute or so and the HM-10 just reconnects by itself but this is also happening frequently.

**To Reproduce**
Enable Telemetry under Hardware > Bluetooth
Connect with Telemetry Viewer or a HM-10

**Expected behavior**
A stable connection until i disconnect by myself

**General information**
 - OpenTX version 2.3.9
 - Hardware: Frsky X-Lite and Full Crossfire Module

skaman82, 2020-06-24:
closing this for now as I stumbled on the cause probably. have to check first

skaman82, 2020-06-24:
Okay, this issue is only happening if the hardware telemetry is set to "Max baud 400000". If it is set to "max baud 115200" the BT connection is stable. But I would like to use my CRSF on full speed of corse, so I am reopening this

3djc, 2020-06-24:
Isn't that exceding BT bandwidth ?

skaman82, 2020-06-24:
I thought opentx somehow handles the conversion between CRSF native baud rate and BT output baud rate as the BT is of course not capable of transmitting at native CRSF baud rate

3djc, 2020-06-24:
We adapt rate, but if there is too much data, well, there is not much we can do about that

skaman82, 2020-06-24:
Hm, any chance to filter some data out without restarting the connection again and again?

3djc, 2020-06-24:
Not sure how we could decide what shoud go or not go

skaman82, 2020-06-24:
maybe selectable setting to do A. all sensors or B. just battery and data relevant for navigation only?

skaman82, 2020-06-24:
Same could be also handy on the "Telemetry Mirror on serial port" feature on radios like X9D. As far I know there is no option to change internal telemetry rate on X9D like on the x-lite

kilrah, 2020-06-24:
We don't exactly want to start adding custom processing and filtering for specific protocols.
Can't you reduce the sending rate of whatever sensors you're using?

skaman82, 2020-06-24:
@kilrah I was not aware of this option in openTX. Where i can find it?

skaman82, 2020-06-24:
Access modules (like R9) have probably the same issue as they are working on a very fast baud rate as well.

kilrah, 2020-06-24:
> About "sending rate of whatever sensors": I was not aware of this option in openTX,

OpenTX can't do anything about it, but it could be configurable in your sensor, or whatever the onboard device sending telemetry data is. 

> You don't need to do filtering for specific protocols. Maybe adding custom update/forward rate to every telemetry sensor would be a good solution like: "native, 0.1 / 0.2 / 0.5 / 1.0s

Still requires us to buffer incoming messages, compare them to a list of different behaviors to act accordingly, make some kind of UI to configure it... the current system is "byte comes in, byte is pushed out". 

> Access modules (like R9) have most likely the same issue as they are working on a very fast baud rate as well (PXX2 is 450k I think). 

They don't, link baudrate between radio and module is irrelevant, it's the quantity of data that can go over the telemetry link, and it's lower on all other systems.

arf-it, 2020-08-16:
> > About "sending rate of whatever sensors": I was not aware of this option in openTX,
> 
> OpenTX can't do anything about it, but it might be configurable in your sensor, or whatever the onboard device sending telemetry data is.
> 
> > You don't need to do filtering for specific protocols. Maybe adding custom update/forward rate to every telemetry sensor would be a good solution like: "native, 0.1 / 0.2 / 0.5 / 1.0s
> 
> Still requires us to buffer incoming messages, compare them to a list of different behaviors to act accordingly, make some kind of UI to configure it... the current system is "byte comes in, byte is pushed out".
> 
> > Access modules (like R9) have most likely the same issue as they are working on a very fast baud rate as well (PXX2 is 450k I think).
> 
> They don't, link baudrate between radio and module is irrelevant, it's the quantity of data that can go over the telemetry link, and it's lower on all other systems.

@kilrah this sounds really logic. I could be wrong but looking at the code the BT module speed is set at 115k (it looks hard coded to me but I could be wrong) despite the telemetry mirror speed set in the hardware menu.
When I'm using the crossfire I set the speed at 400k in the menu. With this configuration  I'm pretty sure that the phisical sport of my Taranis is running at the same speed (400K) while Bluetooth is at 115K and I'm experiencing the same issues reported by @skaman82.
It could be that the BT UART its just overflowed?

3djc, 2020-08-17:
Yes BT is clearly is overflow by CRSF telemetry rate, is CRSF is running at 400k.
Things we can do short term:
- allow users to reduce sport max baurate even on radio that always allow 400k (and therefore are not showing the option)
- allow faster mirroring on devices with external HW uarts (TX16S)  (it does 115k there as well I think, need to check)

arf-it, 2020-08-17:
@3djc  I agree in principle with the first option for my Taranis (that, by the way, need the inverter mod to run at 400K..). I personally don't see the need to have telemetry running at 400K: I guess that data coming down from the flight controller is not faster than 2~5 Hz but I think that the same port is used by the CRSF module to exchange data with the Taranis and reducing to 115K probably will affect performance somehow.
I can already use that option (I see 115K in the hardware menu) and in this case, I'm experiencing a "sensor lost" warning time to time (this will never happen with speed at 400K). This is just to say that approaching this way probably will create lot of troubles to people not using the telemetry mirror.

Long term I think could be great to "decouple" the telemetry mirror feature from the main communication speed. How to do it: no idea but I saw a project running LTM telemetry on serial via LUA script could be a good idea to give access to Bluetooth to LUA in order to mirror telemetry via LUA (there is already an issue opened on this topic) and maybe buildup LUA libraries that mirror telemetry.

brainbubblersbest, 2020-12-15:
#8073