opentx/opentx #7767: Open TX 2.3.9 issue with sport telemetry update

robertha64, 2020-06-25:
After flashing OpenTX version 2.3.9 to Radiomaster TX16S, all the sensors from Flight Controller (Omnibus F4V3 Inav) has random issues with updating (updating few times, then no updating about 20seconds and again updating and so on) and alarm "sensor lost" appears periodically. Tried flashing internal module with latest multimodule FW (1.3.1.9) - no ifluence - same behavior.

When flashed back opentx-tx16s-Radiomaster (7ee5b1d0) supplied with unit, the problem disappear and all sensor update as expected.

3djc, 2020-06-25:
I happen to work with the same radio and FC, no problem here whatsoever.  What radio protocol ?

robertha64, 2020-06-25:
Frsky LBT 8ch

robertha64, 2020-06-25:
Flashed 2.3.9 from SD card, power cycle.
Same issue, few seconds updating all sensors, but then sensors from FC start loosing update and after about 1 minute I get the sensor lost alarm.

robertha64, 2020-06-25:
Hm, weird after about 8 minutes its starts updating completely ?

robertha64, 2020-06-25:
Yes, tuned to highest signal as described and TX is about 5 meters from RX. I have also Yaapu mapping Widget installed (and working when telemetry updates GPS).

3djc, 2020-06-25:
well you want to try to narrow down thngs a little. Please create a new  model with just default setup, bind, discover sensor and check what is happening

robertha64, 2020-06-25:
Hm, not succeed to upload pic

robertha64, 2020-06-25:
[image]
[image]

3djc, 2020-06-25:
your rxbt is really low, is that intended ?

robertha64, 2020-06-25:
No, can be FC issue as RX is powered from it?

3djc, 2020-06-25:
not sure, but this is at least an intrigingly low rxbt

robertha64, 2020-06-25:
Find out - I did supply 5.2V on vbat connection of the FC, so internal regulator strugles to keep higher voltage on internal 5V rail. Now suppling 3S on vbat and rxbt is 5V. 

By the way the sensor update issue dissapear (before I change supply voltage), so I will probably close this as some magic temporary unknown glitch ?

closed 2020-06-25 by 3djc

robertha64, 2020-06-25:
OK thank you for your help.

diogo-f, 2020-09-17:
Today something similar to this happened with me. I'm using a Jumper T18 (with v2.3.8 officially adapted by jumper) and TBS Crossfire Micro TX V2 with firmware v3.72. 

I was flying and had a small crash, then I checked the TX screen and noticed the telemetry info on the TX was not updated with the info on my goggles. 
I rebooted the drone and when it connected it was still displaying the same old telemetry for a few seconds!! So this seems to be on the TX side, but if its happening randomly with other transmitters it seems to be on the radio and opentx... 

let me know if there is anything I can do/send to help, does opentx write logs on the sd card?

(I came to opentx to see if there is a new version with [t18 support and it will be in 2.3.10](https://www.open-tx.org/radios.html), I also have a new firmware for the Crossfire TX but some people are having issues with the new v4 so i'm waiting for that one to be stable.)

3djc, 2020-09-17:
Just means you had lot the crsf link thats all