opentx/opentx #7770: Missing sounds for "Timer X elapsed" in spanish sound files

ignaciobfp, 2020-06-26:
Sorry to make this a bug report, but I was unable to find where I can submit new sounds for the SDCard.

AFAIK there is a sound speaker/recorder, but I am not familiar with this project and I have no idea if this is related to this. In short, sound files for timer elapsed are missing in spanish, so only a beep sounds.

Path for this files would be as follows
SD_ROOT\SOUNDS\es\SYSTEM\timovr1.wav
SD_ROOT\SOUNDS\es\SYSTEM\timovr2.wav
SD_ROOT\SOUNDS\es\SYSTEM\timovr3.wav

It was a bit annoying for me so I prepared some valid wav files for my own usage, but I have no idea on how/where I can submit them to be included in future sdcard content releases. Am I missing something?

HW: Horus X12s, SW: OpenTX 2.3.9

kilrah, 2020-06-27:
SD card contents are autogenerated by a script based on this code: https://github.com/opentx/opentx/blob/2.3/radio/util/tts_es.py

If you tell us the words it should say we can add them.

ignaciobfp, 2020-06-27:
> 
> 
> SD card contents are autogenerated by a script based on this code: https://github.com/opentx/opentx/blob/2.3/radio/util/tts_es.py
> 
> If you tell us the words it should say we can add them.

Thank you for pointing me in the right direction and for providing the information, I will try to add a pull request myself to add the missing sounds.

PR #7777 (2020-07-04): Add missing tts sounds in spanish, minor fixes
Add missing tts spanish sounds (fixes #7770), minor improvements to make some lines more accurate (most cases) or just more natural
commit 3da92560 (2020-07-04): Add missing tts sounds in spanish, minor fixes (#7777)

Add missing tts spanish sounds (fixes #7770), minor improvements to make some lines more accurate (most cases) or just more natural
closed 2020-06-27 by ignaciobfp