opentx/opentx #7772: Crash when you select others model at 2.3.0

JoshuaShaw, 2020-06-27:
Here is the bug video.
[http://easyexcelif.com/images/bug.01.gif](url)

JoshuaShaw, 2020-06-27:
[http://easyexcelif.com/images/bug.01.gif](http://easyexcelif.com/images/bug.01.gif)

closed 2020-07-07 by bsongis

AlexDaniel, 2020-09-14:
Sorry, but I have the same issue, I think, and it is not resolved. In fact, my 9XR Pro has been bricked for quite a while because of this. I tried updating to a nightly version which I believe has https://github.com/opentx/opentx/pull/7775 but it does not fix the issue.

Basically, I was looking through the settings and had an unfortunate idea to select XJT in External RF. Now my transmitter is essentially bricked. If I try to change XJT to something else it immediately crashes and the setting remains. If I try to select another model then it crashes too. I can't figure out a way to get out of this situation. Connecting the transmitter to PC never really worked for me, so I can't change the setting from there. What do I do?

pfeerick, 2021-03-20:
This is indeed not resolved - 2.3.11 still has the same 'instant crash' when choosing XJT as the external module, or changing from it if you have it configured already in a memory. This is similar to #7594.

BrettRD, 2021-05-07:
This is not fixed in 2.3.11.
The only way out I found was to use companion to delete the model.

In my case, when I opened it in companion, the offending model had red text in the RX# column. Companion complained something about "source - ???" when I tried to write back to the radio after selecting PPM.
I can't seem to replicate the problem companion couldn't solve, but maybe that symptom is meaningful (corrupted data structure?).

Fiddling with this I was able to also trigger https://github.com/opentx/opentx/issues/7594