opentx/opentx #7773: I can not flash firmware to rx from 9xr pro

JoshuaShaw, 2020-06-27:
there are no the option of flash firmware to rx in sd-card menu.
[Gif of the bug](https://easyexcelif.com/images/bug.03.gif)

dgatf, 2020-06-27:
You must flash the firmware in bootloader mode (trims inwards while powering up the tx)

kilrah, 2020-06-27:
He says flash a receiver.  Was never supported on 9XR-Pro et al.

JoshuaShaw, 2020-06-28:
Is there any plan to support it?

lucajon, 2020-08-17:
hi, i got the same problem, turnigy 9xrpro opentx 2.2.3. no option to flash firmware of the receiver(xm+) and xjt module. i remember that i have done it before (2017) with this radio. 
I don't remember which version I used... have you find the fix?

lucajon, 2020-08-17:
'normaly' with the file .frk in the firmware folder, when you select the file you should have the flash rx menu option! but nothing!?!

closed 2021-06-03 by stale[bot]