opentx/opentx #7782: Add ability to reorder LUA scripts in Tools menu

jyunte, 2020-06-30:
I searched for this feature request and couldn't find it.

When adding or updating a LUA script, Opentx places the new script at the bottom of the list in the Tools menu (long-press Menu).

I'd like to be able to reorder the scripts using the same method as reordering models... Long-press enter to open a menu to Move, Copy, Delete.

kreidler, 2020-12-06:
Same here.
Just updated the BF Luas to actual version. Have had this script as a telemetry screen. Now changed to the bottom of the TOOLS menu. Even if you delete all luas in tools dir, do a power cycle and copy the file back the sorting order is different but without any sense. 
I am not sure if "copy" and "delete" will be helpful. A single option to reorder the single entries would be sufficient.

JimB40, 2020-12-18:
Adding same request. It seems now TOOLS folder files are sorted by file's modification date. So we have to scroll to most used scripts.
Simple sorting by file name or toolName inside lua (toolName = "TNS|name|TNE") will do the trick.

ichiemo, 2021-05-01:
I 4th this, another way could be renaming/adding a number value to the beginning of the lua script file/folder to order them in the menu.

closed 2022-04-16 by stale[bot]

JimB40, 2022-04-24:
@pfeerick I thinlk we haven't implemented alphabethic sorting. Re-open or we leave it close as in new UI there will be new App section?

pfeerick, 2022-04-24:
ROBERT! You need more coffee! ;) 

1) Why would I re-open something on a project I'm not a member of?
2) Look again, "we" did, and it's been like that since 2.6, so only been alphabetically sorted for the last five months! 😆

JimB40, 2022-04-24:
Well for years working with devs I learned that when they say "It's done" it means coding without testing :)

EdgeTX 2.7

[image]
[image]

pfeerick, 2022-04-24:
No, that means it was implemented in colorlcd - you didn't ask *where* it was implemented ;)

JimB40, 2022-04-24:
Buahaha. I love devs