opentx/opentx #7785: 2.3.9 with 9XR Pro and Multimodule: menu entry not visibale

martin-at-schorsch, 2020-07-02:
In the Module Setup on Turnigy 9XR Pro with 2.3.9 some menu entry are blank or will be blank when scrolling in Type or Subtype.
Also "Bind" is not visible or selectable.

**To Reproduce**
Error is reproducable on Hardware as well as in the Radio Simulator.

Steps to reproduce the behavior:
1. Go to 'SETUP' Page 2 and scroll down to External RF and select in "Mode" "MULTI".
Scroll in "Type".  After "MJXq" type vanish. 
Workaround to go back is to select something else than MULTI in Mode and set MULTI again.

Bind is also not visible&useable for other Modes like DSM2.

**General information**
 - OpenTX version 2.3.9
 - Hardware 9X Pro with Jumper 4in1 Multi Module

martin-at-schorsch, 2020-07-02:
maybe related to #7772 #7775 ?

pfeerick, 2020-07-10:
I just realised that the bind, receiver ID, etc options are missing from my 9XR-PRO with 2.3.9... which were present on the 2.2.x firmware series. IIRC, there is a blank line where it would have been, and there is also a 'sync' menu entry which a) I don't know what that's about and b) doesn't do anything. Exactly as shown in the simulator. 

[image]

And somewhat similar to #7772, if I try to change model, try to create a new one, or change the RF module (mode) ... instant crash and reboot. There is also a exclamation mark up the top right on the main screen... not sure what that's about! I can't remember if I had the EEPROM write progress bar timer option set in the past on the 9XR-PRO, but that is roughly where I remember seeing it on either the 9XR-PRO or the 9XR. And the final straw is that the SD card isn't being read - and just reformatted another card as FAT16 and loaded the `sdcard-128x64-nolua-2.3V0029.zip` from https://downloads.open-tx.org/2.3/release/sdcard/opentx-9xrpro ... with no success. I think I'm going to have to go back to v2.2.x and forgo the USB simulator feature as 2.3.x has some serious bugs making normal operation of the transmitter impossible.

lucajon, 2020-08-20:
same thing for me (9xrpro) when you try to create new model it crash.
pfeerick : do you have a version where you have the menu to flash the receiver or the external module working?
mine (2.2.4) no menu to flash my xm+ receiver...

khttkelite, 2020-09-13:
Any fix for this? Same thing happened to my radio when i updated to 2.3.9. Works fine with first flash but when i select a module or create new model it crashes. After that after booting splash screen does not show up, sd card is not visible and can't make a new model or change the module type in settings.

3djc, 2020-09-13:
There is unfortunatly not a lot we can do as none of the devs has access to this type of radio anymore

pfeerick, 2020-09-13:
Does the fact that the radio simulator also exhibits the same issues as the hardware not help in debugging it?

pfeerick, 2020-09-13:
@lucajon I don't use XM+'s / radio flashed receivers, so I can't help you with that sorry.

pfeerick, 2021-03-20:
This issue can be closed as the core issue was resolved in a later update. 2.3.11 has the missing menu entries, and the bind button is visible (on the Turnigy 9XR-PRO). 

@lucajon
> pfeerick : do you have a version where you have the menu to flash the receiver or the external module working?
> mine (2.2.4) no menu to flash my xm+ receiver...

I was just getting my Taranis up and running, and saw the entries for updating receivers there (radio setup -> SD browser -> select a receiver firmware and update external device), so thought I'd look at the 9XR Pro... it doesn't look like it was ever supported via OpenTx on this hardware? It is however supported in the 'Maintenance mode' of erskytx, so that could be a clunky way of doing it via your transmitter  as unfortunately whilst the newer 3.0 of the bootloader (which you can also get from the erskytx binary, and seemingly use with opentx since they both look to use the OpenRcBootloader - just opentx uses the last tagged release, rather than a much later version) allows for running apps that will let you update the multi-protocol module, changed sport IDs, etc, sport updating doesn't look to be an app :(

closed 2022-04-17 by stale[bot]