opentx/opentx #7791: Add ability to map all 16 channels in Companion/Radio Settings/Trianer

dwheel, 2020-07-04:
When using an MPM as a receiver for wireless training purposes, all 16 channels from the student transmitter can be utilized in the instructor transmitter. However, Companion only allows the first 4 channels to be mapped. Please add the ability to use all 16 channels in Companion.

kilrah, 2020-07-04:
You use the TR1 to TR16 sources in the mixer.

dwheel, 2020-07-04:
Yes, I know. But the control sticks in the companion trainer simulator are always mapped to channels 1 thru 4. They can not be used for simulation if someone wants to use any of the other channels of the student transmitter for A,E,T,R. 

[image]

kilrah, 2020-07-04:
Ah, completely different thing.

closed 2021-06-04 by stale[bot]