opentx/opentx #7794: Throttle Warning for a Centered Throttle

bigtwin, 2020-07-05:
Cars and Ships are usually capable to move backwards and the Zero-Gas Position is at neutral position. 
It would be usefull to have a _Throttle Warning_ for throttle not being in center/neutral position.
I think of something like a check box named _Centered Throttle_ at the _Model Setup Screen_ right beside the _Reverse Throttle_ check mark.

Especially if you have airplane models on the same transmittter and you change model from an Airplane to e.g. a Ship,  it happens rather often that the Throttle is way down which is usually zero for planes. It would be great to have a _Throttle Warning_ in that case.

bigtwin, 2021-01-02:
I still treat this as a helpfull feature. If I change a model from an airplane to a ship, it would help if I get a _Throttle Warning_ if the throttle stick is not in center position (which is usually zero on a ship). Therefore it would be helpfull if I can choose if _Throttle Warning_ is be triggered at 0% rather than -100%.

drtrigon, 2021-02-11:
I also see this as a very basic and needed feature as the requester! I would propose a field next to throttle warning that allows to specify which value it wants to see. That value would be -100% by default and we car RC guys could then change it to 0%. Of course it should allow a range around that value say +/-5% to be satisfied.

If you tell me where to start I could offer to try to modifiy the code and push a pull request.

bigtwin, 2021-02-11:
Any comment of the development team would be appreciated? Its now more than half a year in queue ... 
Is there any hope for this feature to be considered?

drtrigon, 2021-02-13:
Also I could image that acro/3D heli (with negative pitch and v-curve) and position/loiter mode drone pilots might want other throttle defaults than -100%.
For example -50% or 0%.

drtrigon, 2021-02-13:
And from "marketing" point of view this would be nice one, as it would pop up right prominently on the first model config page... ;)

drtrigon, 2021-02-14:
It might may be help to use to given form for this feature request - may be that would make the development team more willing to respond to this?

**Is your feature request related to a problem? Please describe.**

Cars and Ships are usually capable to move backwards and Zero-Gas is at neutral position. It would be usefull to have a Throttle Warning for throttle not being in center/neutral position.
Also acro/3D heli (with negative pitch and v-curve) and position/loiter mode drone pilots might want other throttle defaults than -100%.

Especially if you have airplane models on the same transmittter and you change model from an Airplane to e.g. a Ship, it happens rather often that the Throttle is way down which is usually zero for planes. It would be great to have a Throttle Warning in that case.

**Describe the solution you'd like**

I would propose a field next to throttle warning that allows to specify which value it wants to see. That value would be -100% by default and we car/ship RC people could then change it to 0%. Of course it should allow a range around that value say +/-5% to be satisfied.

**Describe alternatives you've considered**

I think of something like a check box named Centered Throttle at the Model Setup Screen right beside the Reverse Throttle check mark.

**Additional context**

(none)

drtrigon, 2021-03-16:
ER9X has it e.g.: https://littlewingaero.wordpress.com/tutorials/turnigy-9x-er9x/er9x-throttle-safety-switch/

Using a [Logical Switch and a Special Function](https://www.openrcforums.com/forum/viewtopic.php?f=45&t=11850) would allow me to trigger a one-time voice message on model load if throttle is not ~0. That is ok, but I don't know how to make the message repeat until throttle is ~0 and I don't know how to display a warning message (like the other pre-flight checks do)?

I also looked into [Lua Scripts](https://doc.open-tx.org/opentx-2-2-lua-reference-guide/part_i_-_script_type_overview/function_scripts#advanced-example-save-as-scripts-functions-cntdwn-lua), but I do not understand how to make [popupWarning](https://doc.open-tx.org/opentx-2-2-lua-reference-guide/part_iii_-_opentx_lua_api_reference/general_functions/popupwarning) disappear as soon as throttle ~0?

drtrigon, 2021-04-27:
Use 2 logical switches and 1 special function:
- make a logical switch (e.g. L01) that checks for |delta x| < 10 (or any other value you find convenient)
- make a logical switch (e.g. L02) that does an and on both switches (e.g. AND L01 L02) and add a delay of e.g. 0.5s
- make a special function that checks for the 2nd switch NOT to be active (!L02) and then plays either the default throttle warning or I let me just read the Thr value in intervals of 2s as that allows me to realise to go to center instead of bottom

I think the first logical switch and the special function are self-explaining. The 2nd switch (L02) is a memory that stores whether the stick was in correct position at least once and for at least 0.5s.

Now what I still miss is how to display a warning message/screen?

bigtwin, 2021-11-15:
Sorry for not recognizing your post 6 month ago.
Following up on this one ... perfect would be a warning just like the known _Throttle Check_ upon selecting a model - the transmitter does not get hot, before putting throttel to -100% or confirm it.

Meanwhile I helped myself with a solution similar to yours. Throttle-Out is set to 0% as long as the Throttle control was  not moved to the middle position.

Nevertheless, I feel this should be build into OTX.

drtrigon, 2021-11-16:
> Nevertheless, I feel this should be build into OTX.

I fully agree. I also think somebody of the core dev team should comment here what the status is.

bigtwin, 2022-03-05:
Just a push up ...
I still feel that this issue is very helpful and supposably **easy to implement**. 
Just another _check box_ on the _Setup_ screen for a model. If the box ist checked (e.g. _Centered Throttle_), a Throtte Warning will occur when the throttle stick is not in centered position:

[image]

bigtwin, 2022-07-29:
This issue is being perfectly solved by the _EdgeTX_ team ...

drtrigon, 2022-08-18:
Intressting a reason more to move...?

Can you give some details, e.g. a link?

bigtwin, 2022-08-18:
Its introduced in 2.7:
--> https://github.com/EdgeTX/edgetx/pull/1395

bigtwin, 2023-10-20:
As I started this issue, I am closing it now. After 3 years I dont think that it is of relevance for anybody. I migrated to ETX 2..9.1 and it works perfectly there. 
If anyone think it would be helpful for some reason, feel free to reopen it.

closed 2023-10-20 by bigtwin