opentx/opentx #7795: Widget option type STRING doesn't work

wimalopaan, 2020-07-05:
The widget option type STRING is shown and can be edited. But the value of the option is completely garbled.

3djc, 2020-07-05:
Lets see, no details on version, radio used, operation done, expected result and seen result ...

wimalopaan, 2020-07-05:
OpenTX 2.3.10 (master)
RM TX16s

using 
local options = {
  { "Next",     SOURCE, 8},
  { "Previous", SOURCE, 9},
  { "Select",   SOURCE, 10},
  { "Name", STRING, "xxx"} 
}

In companion: everytime I choose the widget setup, I see random characters. I can edit, but reading the option value gets random string.

In radio: in setup I see the correct default string, but still get random chars reading the option.

pH7-JP1, 2020-11-27:
The STRING option does have problems: AFAIK there isn't a way to control the number of characters - the user is allowed to input 8 characters, but only 4 are returned. The value returned in the options array is not a lua string, but a 32 bit integer with 4 byte sized codes as described in the attached function. If an updated string has the same characters with only conversion to upper / lower case of the same characters, then the widget's update function is not called. 

Still, I was able to make it work well enough to implement the widget I was working on by writing the attached function to convert the integer into a 4 character string. 

[getWidgetString.txt](https://github.com/opentx/opentx/files/5609816/getWidgetString.txt)

sashgorokhov, 2021-04-24:
This is really annoying to not have proper STRING handling in opentx. I wish we could type lengthier strings for proper descriptions and stuff.

pH7-JP1, 2021-04-24:
I absolutely agree with sashgorokhov. Even though wimalopaan did not specify, here is the info that 3djc requested: RadioMaster TX16S running OTx 2.3.11. Expected results to have a LUA STRING returned in the options table if STRING is requested. The seen result is that a 32 bit integer is returned in the options when the user enters characters.

closed 2022-04-16 by stale[bot]