opentx/opentx #7796: sbec lua not working in 2.3.9

P-I-Engineer, 2020-07-05:
sbec tool is not working

**To Reproduce**
Steps to reproduce the behavior:
plug in battery
-discover sensors
-go to tools, sbec
-syntax error - no sbec in this model
or syntax error attempt to call a nill value.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version 2.3.9 and 2.3.10nightly
 - Hardware X9D, X9 lite
 - OpenTX Companion 2.3.9
 -FrSky FrESC32 BL_Heli32 80A 3-6s ESC

3djc, 2020-07-06:
Are you certain this model is supported by the scripts ? Went throught the product user manual, and no mention of scripts support on this product.

P-I-Engineer, 2020-07-06:
Just googled and dawned on me as well, no adjustable Bec.

It's either 5v or 12v

Sorry for confusion, thanks for time

closed 2020-07-06 by P-I-Engineer