opentx/opentx #7797: TX16s - Crash in Joystick mode on Mac

CyberMac1, 2020-07-06:
OpenTX ver: 2.3.9 (2.3.9-OTX (60e1edc1))
EEPR: 219

MacBook Pro MacOS 10.15.5

When connecting the TX16x to a Mac and selecting Joystick (HID) the transmitter will turn off after 20 - 30 sec. 
(only the 6P button selector will stay on)
Pressing the Power button (after disconnecting the USB cable) result in an Emergency Mode.. 
Power off and back on resolve the issue until next try. 

As the MacBook only have USB-C connectors, I initially connected the TX16s using a USB 3.1 adaptor with Genesys Logic, Inc. chipset. this results in a TX16s crash after 20 sec. 
Changing the Adaptor to VIA Labs, Inc. chipset results in much better stability of around 20-30 minutes until system crash. 

Both adaptors are working in 480 Mb/s

GotskGK, 2020-07-11:
That's quite odd as I'm running it with Mac as well with the latest 2.3.9 and have had no issues.

CyberMac1, 2020-07-11:
Apparently this is an issue with the TX16s top board. RadioMaster contacted me and they are sending a new top board.

closed 2020-07-11 by CyberMac1

apealphaghost, 2020-07-27:
anyone else get white blank screen and no recognition from opentx