opentx/opentx #7798: Different output on telemetry Mirror with f.port

brainbubblersbest, 2020-07-06:
Hello, 
I dont know if there is a Bug or already documented. 
But I dont found Information about this. 
Even in the official f. Port Documentation. 
If I use f.Port, and Trace the data stream over a bt connection to a device that need gps Position and altitude over S.port, it cant work with this data. 
Data is received and inverted. This is all checked with a two channel scope. And counterchecked with normal s.port receiver firmware. 

Is there any Information available that f.port affect the data stream on bt mirror? 
And in which way it does? 
I thought Till today. F.port affect only the communication between RX and FC. 
Hardware:
X12s
Hm-10
S.port inverter to uninvert the signal
Mfd crossbow Mini AAT

kilrah, 2020-07-06:
We are not aware of any difference on our side.

brainbubblersbest, 2020-07-06:
I will try to make a Video of the two setups.
The Baudrate of the data came out of the HM-10  is not different in any way?
I use 119200 for HM-10 to internal BT
And 57600 for a connection between to hc-05 modules. 
But here i use the sport connector on r9m. 
This is btw only the configured Speed in the BT Modules .
But is that baudrate also correct for the data that goes trough this BT Connection? "used at BT Mirror"?

brainbubblersbest, 2020-07-07:
Now im totally confused. 😅
I need uninverted s.port on the tracker. 
Left picture Shows uninverted s.port at top, and s.port came out of the hm-10. Nice and symetrical.

Right picture Shows uninverted s.port at top (with other time setting) and s.port measured at the back of the r9m.
Maybe not symetrical in relation to the transmition lag?
Or related to wrong baudrate setting? 

But the s.port at back of r9m has the same Signal direction? Wtf? Any Suggestions?

This is same behavior with s.port and f.port setup. 🤔

brainbubblersbest, 2020-07-07:
[image]
[image]

brainbubblersbest, 2020-07-19:
I figured out, that on s.port i have the same behavior. Updated Opentx before i noticed this issue with the internal telmirror. 
But the last working OTX Version i get usable Telemetrie Outputs where 2.3.3. 
There where a lot of Issues out there related more or less to BT and Telemetry Mirror.
some are closed without an solution ?
https://github.com/opentx/opentx/issues/7314
https://github.com/opentx/opentx/issues/7192
https://github.com/opentx/opentx/issues/7829
https://github.com/opentx/opentx/issues/7766
https://github.com/opentx/opentx/issues/7631
https://github.com/raul-ortega/u360gts/issues/91
https://github.com/opentx/opentx/issues/7799

kilrah, 2020-07-19:
There are so many issues with BT on these radios that we've mostly given up.
FrSky have 3 different BT systems that are incompatible with each other, within one system there are multiple firmwares on the modules that are out there with different bugs, and FrSky have never distributed updates to allow putting everyone to the same level.

brainbubblersbest, 2020-07-19:
Cant this be done with an opt in tick. Like internal access?

Like Holding the old Code when a non express Radio is used and the bt
module is not Upgrade?
The BT has worked not bad for telemetry use in 2.3.4.
And in 2.3.3 are made some improvements to bt already, if I remember right.
The ability to connect a bt module to the internal aux 1 (p13) is disabled
on Horus Transmitters.

But on radio Master and Jumper, the function to use a bt at aux1 or aux2
will developed, if I read that right, with Horus x12s BT modules too?
Correct me when im wrong.

The changes to the bt Code should be reverted to 2.3.4 state to got it
working again.
If the module is Upgrade with the Para one. That can be explicit activate
on flashing otx. Thats just an idea.

If a Tool exists to configure the BT Firmware. The original BT Firmware is
available at frsky x12s Flash contents. And its flashable trough the Horus
Bootloader. (BTLEmodule.bin)
Maybe thats Helpfull in any way.

bsongis, 2020-10-08:
Andreas just asked me to look at this issue, it's very difficult for me to understand.
S.Port is serial. Is it possible you attach here a logic analyser trace from what you receive so that I can check if there is an issue?

brainbubblersbest, 2020-10-08:
I got working telemetry if I pull the Signal from the s.port pin in the Jrbay. OR the s.port pin of the r9Module.
Rxsr and r9mm and also a r9slim+ are configured as SBUS/S.Port. 
I finally dont use F.port because the Tracker seems to dont understand the different sensor ids. 

But If I connect a HM-10 or a HM-18 to the internal Bt Module: I get some Data, but no valid Telemetry Data.
I tried with:
-Baud settings 115200 and 57600 at the Hm Modules.
-every available Firmware for the Modules.
-BT is set to Telemetry in Horus and 
-the Mac Adress is recognized in OpenTX

This has definately worked with 2.3.3 and in the Tracker Community are multiple Users with this Issue.
Additional to that: the Frsky Free link App also receive no more valid Data. 

I dont have a logic Analyzer, but I can get one and try to make some measurements, if this is helpfull to fix that.

brainbubblersbest, 2020-10-11:
[TelemetrieStream from internal BT Mirror over HM-10]
(https://github.com/opentx/opentx/files/5361882/capture.txt)

This is the data received from internal Telemetrie Mirror.
with a bound HM-10. Configured 115200Baud

-Receiver is a R-XSR Connected to F722-Wing, configured in SBUS/S.Port

-Telemetrie on Horus is received. Tracker understand nothing.

[image]
[image]
[image]

brainbubblersbest, 2020-10-11:
And this is where i get valid Telemetry data. the Trackershow me the Coordinates Received by R9M over a R9MM

[Sportoutput from R9M Back Connector .txt](https://github.com/opentx/opentx/files/5362338/capture.txt)
But this works with the Method above under 2.3.3 too, so here must be something different.

[image]
[image]

brainbubblersbest, 2020-10-12:
How is this related to this issue? 
[RCGROUPSPost](https://www.rcgroups.com/forums/showpost.php?p=45564757&postcount=3450)

brainbubblersbest, 2020-10-12:
The only difference i can See here is that:
In the valid stream direct out of the r9m the stream always contains a 7E followed by an 7E.  (7E7E)

The not valid stream out of the BT Mirror contains only one. 
So the last two bytes are dropped. 
Maybe it's related to #7874  But that issue is about aux Serial Mirroring.
Is that helpfull for you @bsongis?
And maybe it's the reason why Free link also is not working since 2.3.3.

brainbubblersbest, 2020-10-12:
[BT Mirror vs S.Port.txt](https://github.com/opentx/opentx/files/5366800/BT.Mirror.vs.S.Port.txt)

brainbubblersbest, 2020-11-01:
I Made a mistake at the described otx Version.
Working otx was 2.2.3 maybe 2.2.4
I will try 2.2.4 also and make a hex dump of the received data.

Narnasqueneth, 2021-01-21:
following. I also have r9m (pre 2019 version) and am trying to get it to work with the MFD mini crossbow tracker. I am using f.port though...

closed 2022-04-17 by stale[bot]