opentx/opentx #7799: No telemtry Data on TX16S with CF with 2.3.9

bswoolf, 2020-07-06:
I use a TX16S with a HX100 und Crossfire Micro tx. If I use Firmware OpenTX 2.3.6 or so (the Firmware it ist on it when delivery without any companion support) CF-Telemetrie works fine, after Upgrade to 2.3.9 Telemetrie is connected but the are no Data’s at all, either BF Data nor CFown Data. But it isconnected and they are shown stars on last Site of Modell Menü.

3djc, 2020-07-06:
As told on RCG, it is an issue with you specific configuration, as there a many TX16S/CRSF users out there, and none reports anything like you. Also, not sure of what firmware you are referencing since the one delivered with TX16S does not have a version, just a date (4/28)

bswoolf, 2020-07-06:
2.3.9-otx(60e1edc1) crossfire, faimode, internalmulti, multimodule, lua, luac
No Data, but telemtry is connected

opentx-tx16-Radiomaster (7ee5b1d0) crossfire, internalmulti, multimodule, luac   (firmware_tx16s_1_en_200428.bin)
Telemtry with Data, everything OK

Is this usefull or need further Information?  Is it possible, that the rx in HX100 is the Problem? It was a rtf buy. 
I made one Change, VTX was on TX2 made from Betafpv and i changed it to TX 1, because TX2/RX2 is for Receiver.

kilrah, 2020-07-06:
> I made one Change, VTX was on TX2 made from Betafpv and i changed it to TX 1, because TX2/RX2 is for Receiver.

Then you need to update the quad's configuration to reflect that...

closed 2020-07-06 by kilrah