opentx/opentx #7806: Companion opens OTX file before radio profile is selected, causing conversion prompt

benlye, 2020-07-09:
With radio profile section enabled, when launching Companion by opening an OTX file, the file is loaded before the 'Select Profile' prompt is shown.  This means that if the file does not match the radio from the last-used profile there is a prompt to convert it (which is most likely unnecessary).

If the conversion prompt is ignored for a few seconds, both dialogs appear at the same time, with the select profile prompt on top of the conversion prompt.

[image]

The prompt to select the radio profile should come before the file is opened so that the correct radio profile could be loaded before the file is parsed.  The conversion prompt would only appear if a radio profile is selected which does not match the file.

An advanced maneuver might be automatically selecting the radio profile based on the OTX file, but I don't know if that's feasible.  E.g. if the OTX file is for a TX16S, and there's only one TX16S profile, select it.

I'm happy to have a swing at this myself, but am guessing that someone who knows the code would resolve it much faster :-)

3djc, 2020-07-09:
Opening it like that is plagued with issues, not only this one I'm afraid, but yes asking first would be nice. @elecpower, could you help with that ?

elecpower, 2020-07-09:
Likely related to the async method the code works on start up to enable version upgrades to be checked without holding up the interface. This also meant the profile prompt had to be implemented as async to avoid a refactor. There are timer delays but from memory they are a bit arbitrary. Anyway let me have a look.

benlye, 2020-07-09:
It seems to me - without having looked deeply - that the conversion prompt is happening before the async stuff, and that's when the select profile prompt should be.  

I think it would be better to ask for the profile before the UI loads as it's a little confusing that once the UI appears, the last-used profile is already loaded _before_ you ask what profile I want.

One other minor thing - the select profile dialog is resizable when it probably shouldn't be.

elecpower, 2020-07-09:
@benlye as @3djc stated launching Companion with the file name supplied by CLI or via file association is allowed but over time has become very risky. It needs a major code redesign or possibly even safer to ignore the file name so that the environment is set up properly and stable before any actions are taken.
To fix this specific issue quickly it is probably best not to prompt for the profile if a file name is supplied at launch. And yes the last used profile may not match the file being opened  but that is the way it worked before the prompt for profile feature was added or when the option is turned off in the application settings.
I'll work on a PR for the quick fix option.

benlye, 2020-07-15:
I had a look at this myself and offer #7822 as an alternative to @elecpower's PR.

benlye, 2020-07-15:
I also fixed the profile dialog so it can't be resized.

closed 2020-09-04 by elecpower