opentx/opentx #7810: 2.3.9 9xr-pro crash stays in safe mode after reboot

xxxFROSTxxx, 2020-07-10:
When changing model external mode from ppm to xjt and then attempting to change to another mode reboots in safe mode. Unable to clear watchdog error. Safe mode problem persists after recycling power and removing battery with unexp.shtdown error. Turning on transmitter powers on immediately with no boot in safe mode. 

Flashed with generic 2.3.9 as well as heli same problem. 

when attempting to change mode in safemode watchdog reboot occurs guessing because SD card is disabled in safe mode...?

To restore flashed stock 9xr pro image and start over fresh opentx install. Will try nuking the models via USB and see if that fixes the constant safe mode.

kilrah, 2020-07-10:
Duplicate of https://github.com/opentx/opentx/issues/7810

closed 2020-07-10 by kilrah

xxxFROSTxxx, 2020-07-11:
Duplicate of itself?   Can you share duplicate issue number as I have searched for the constant safe mode?

kilrah, 2020-07-11:
https://github.com/opentx/opentx/issues/7594