opentx/opentx #7811: After a few seconds servo output  jumps

gweiller, 2020-07-10:
OpenTX 2.3.7
x9d+  se 2019 
g-rx8 

When the mixer output exeeds some treshold the servo jumps in the opposite direction.  Looks like the sign gets inverted.  So far only seen on channel 8 and  with several seconds delay.  In one situation the event occurred exactly when the output exceeded -128.  (overflow, or signbit problem?).  When the output is reduced, the servo jumps back to correct value.  The effect is only seen after a few seconds and only for negative values.   Don't know whether it is channel specific, but I can output bigger negativ values successfully on other channels. 

Can't reliably reproduce the error yet.  Limiting the output range, or changing the servo does not help.  Perhaps it is an overflow problem in the mixer.

3djc, 2020-07-10:
2.3.7 was pulled back because of an issue. Please update to 2.3.9

gweiller, 2020-07-12:
No change after updating. It might be the same overflow error mentioned in
the notes or a different one.  I guess I report it again as this one is now
closed.

3djc, 2020-07-13:
Then you need a MUCH more detailed report with your OTX and exact procedure to reproduce

closed 2021-01-09 by kilrah