opentx/opentx #7814: Radiomaster TX16S. Companion 2.3.9 downloads wrong mode 2 firmware?

Luznatural, 2020-07-12:
I have two other radios (Frsky X12S, and Frsky X9D+) upgraded successfully to OTX 2.3.9, so I am fully familiar with the overall process.

I bought a brand new Radiomaster TX16S, and after checking that everything was OK with the radio, making several trial flights with an electric plane, I decided to upgrade the firmware to real OpenTx 2.3.9. I selected in Companion the appropiate Radiomaster TX-16S, FCC version, Mode 2, LUA, R9M, order of channels AERT.

Inmediately after upgrading the radio, I noticed that everything went well. except the throttle was vertical in the right gimbal (CH2), instead of the left (Ch3, as it should be in mode 2), and elevator was in the left gimbal (Ch3) instead on the right (Ch2, as it should be in mode 2). Both Ailerons and rudder were in the correct positions (Ch1 horizontal in the right gimbal, and Ch4 horizontal in the left gimbal)

I double-checked that I had downloaded the correct Radiomasted Mode 2 version, and repeated several times the download and the firmwarte update procedures with same results.

I returned the radio to the factory mode 2 firmware, and everything went normal again.

I tried again to do the overall process, with same results. So, apparently Companion downloads a version of OTX 2.3.9 wich is not exactly mode 2 (AERT) as selected, but something like (ATRE).

**General information**
 - OpenTX version (2.3.9, FCC, Mode 2, LUA, R9M, AERT)
 - Hardware (Radiomaster TX16S)
 - OpenTX Companion (2.3.9 for Windows)

I apologize in advance if I did something wrong

3djc, 2020-07-12:
There is no such thing as mode 2 firmware, or aetr firmware. Those are two unrelated concepts. Mode and channel order are both radio settings. If it did change when you updated, it means you did the upgrade wrong. Please go to community sites like rcgroups or openrcforum for guidance, as this site is only for bugs reporting

closed 2020-07-12 by 3djc

Luznatural, 2020-07-12:
Thank you 3djc for closing so fast, without even checking (as you always do).
Probably both concepts are unrelated, as you say, but the problem appears with TX16S and not with my two other radios, and maginally fixes itself, without changing anything, with installing the stock firmware. So, probably is magic, but for sure is not a problem realted with the radio settings.

kilrah, 2020-07-12:
Did you actually check those radio settings?

Luznatural, 2020-07-12:
But of Course!!! several times.
I even tried copying some models from the other radios, and always happened the same, not matter the model.
The first thing always comes to my mind is that I made something wrong, and probably is true also this time, but I cannot find the mistake. This is why I reported it.
Again, I apologize for taking your time.

3djc, 2020-07-12:
Which I why I told you to learn about mode and channel order...

Luznatural, 2020-07-12:
I put channel order AERT in the settings and the radio moves ATRE.
When I install the stock firmware everything returns to AERT. What else I can do?

Luznatural, 2020-07-12:
As adittional information, the internal multimodule shows V1.3.1.9 AETR firmware

kilrah, 2020-07-12:
In radio settings the order only affects new models. For any existing models or those you copy the order is what is defined in the mixer.

Luznatural, 2020-07-13:
Apparently, as far as I check them the mixers are set correctly in the models I copied from the other radios.
Did anyone of you try to reproduce the problem?

kilrah, 2020-07-13:
We don't have enough info for that. What protocol and receiver are you using? Can you see the discrepancy on the channel monitor?

Luznatural, 2020-07-13:
Sorry to hear that you cannot test the problems
Yes the discrepancy appears everyvhere, even in channel monitor. It is coherent throughout the menus.
What is very strange is that the problem vanishes when you install the stock firmware, but appears again installing OTX2.3.9.

kilrah, 2020-07-13:
Post your otx file.

Luznatural, 2020-07-14:
1. This is the stock firmware for mode 2:
[firmware_tx16s_2_en_200428.zip](https://github.com/opentx/opentx/files/4916626/firmware_tx16s_2_en_200428.zip)
2. This is the firmware for mode 2 downloaded from companion:
[opentx-tx16s-flexr9m-lua-en-2.3.9-otx.zip](https://github.com/opentx/opentx/files/4916628/opentx-tx16s-flexr9m-lua-en-2.3.9-otx.zip)

kilrah, 2020-07-14:
I loaded your otx, left it on the selected model (Aero), flashed stock firmware, throttle moves CH3. Flashed the opentx firmware, throttle still moves CH3.

I don't know what you're doing wrong, but it's not an OpenTX problem.

Luznatural, 2020-07-14:
Finally, I was able to fix it, and now all works ok. The only thing I made different this time was flashing the radio directly from the SD card, instead of doing it with Companion.
Thank you all very much for your time, and sorry for the distraction I caused to you.

MuhammadQaisarAli, 2021-09-21:
> Finally, I was able to fix it, and now all works ok. The only thing I made different this time was flashing the radio directly from the SD card, instead of doing it with Companion.
> Thank you all very much for your time, and sorry for the distraction I caused to you.

Thank you, i finds the same problem. let me try your fix.