opentx/opentx #7815: Throttle channel missing a lot of updates when being the first channel (since Open-TX 2.3)

Enurian, 2020-07-12:
**Describe the bug**
While studying my Betaflight-Logs, I was confused why my throttle channel got very "choppy" since I updated my (original) X-Lite to Open-TX 2.3.x (currently 2.3.9 but it was the same with 2.3.2).
By looking at the raw rc command, I could see that it would update most of the time only every 3rd or 4th packet. So, like 20-30ms until the value updated instead of the 6,7ms that would be expected with an R9 system.
Now I got curious and checked another channel: roll was totally fine, updated every 6-7ms. I tested and ruled out a lot of stuff until I changed the channel order.

Normally I´m using TAER1234. I changed that to RAET1234.
Now BOTH axis (throttle and roll) work as expected! When I changed back to TAER1234, throttle was broken again.
My conclusion: This must be a bug.

**To Reproduce**
1. map throttle first, I used "map TAER1234"
2. record the transmitted values. In Betaflight cli:
set rc_smoothing_debug_axis = THROTTLE
set debug_mode = RC_SMOOTHING
3. see the missing packets on the throttle axis in Blackbox Explorer Debug->"raw rc command"
optional: use another channel order like RAET1234 to see the issue gone.

**Expected behavior**
Throttle should work fine when using it as the first channel.

**Screenshots**
Problem can be seen here: TAER1234 throttle channel is choppy
[image]

Fix by using RAET1234 channel order
[image]

Roll is not effected when being first channel (still using RAET1234)
[image]

**General information**
 - OpenTX 2.3.9
 - FrSky X-Lite (both internal D16 and external R9M Lite Pro)
 - OpenTX Companion 2.3.9

3djc, 2020-07-13:
1) you say this problem was nopt in 2.2.x. Do you mean by that you tested 2.2.4 today ? Or was it ok a couple of monthes ago ?

2) This could be caused either by OpenTX, FrSky R9, or Betaflight. We do not treat any channel specificaly at all, so while a bug is always possible, not sure OTX would be my prime suspect for this.

So the best we can do is to confirm we are sending the very same pulses as 2.2.4 to the R9 module, what happens after that is no more OpenTX.  To do that, we are going to have to build a specific version of 2.2.4 and 2.3.9, a lot of work is involved, so please double/tripple check on your side that 2.2.4 is indeed not doing it when use with same same module, receiver and betaflight version as your 2.3.9 tests

Enurian, 2020-07-13:
1. Unfortunately, I cannot test 2.2.4 anymore, since I´m using ACCESS only as of today. It is not supported in 2.2.4, correct? I could pinpoint the switch from 2.2.4 to 2.3.2 as the cause because I looked through old logs and dates from firmware files. I did not change anything else along with the OpenTX Update.

2. I know this could be difficult to locate. That´s why I skimmed through all logs I have. Turns out that OpenTX was the only change right before it started. Settings on the quad were not touched at all. Both D16 (at the time I still had one) and R9 showed the same problem. Later FrSky Firmwareupdates and the switch von ACCST to ACCESS didn´t change anything. Betaflight 4.0.6 and 4.2.0 - no difference. In OpenTX I tested with ADC Filter on and off as well (since I do not recall when I switched it off).

If it is the only way, I guess I could get out my old R9M Lite (non-pro) module that still supports ACCST, then flash back the module, one quad and then OpenTX 2.2.4. I can do this later today, but as stated I´m already pretty sure how the result will look.

kilrah, 2020-07-13:
Also please mention exactly what firmware you're running on the module and receiver, and receiver type.

3djc, 2020-07-13:
What you are suggesting is the first channel is sent last often, so could you make a model where the two first channel are identical and test ? If you see a difference between the two first, then test become simpler because we then just have to check with a similar setup, we are always sending the same value to the modules for the first two channels

Enurian, 2020-07-13:
@3djc Thanks, great suggestion!
At first I have to apologize: I have confused the "R" from "Rudder" with "Roll". In fact, it seems that it is **always the first channel** that is affected.
I tested with RAET1234 because it would cause problems to use throttle when arming (minimum for throttle = hard left for another channel).

So, I mixed my input for rudder (yaw) on the first channel also to aileron (roll) on the second channel. Result: **Only the first channel is affected, the second output is fine**!

Here debug for the first channel (yaw):
[image]

Here debug for the second channel (roll):
[image]

@kilrah 
I haven´t stated these because the issue persistent through a lot of firmwares, modules and protocols. I used the internal module with D16 prior and after the 2.0.1 updates on ACCST, I used the old R9M Lite with both ACCST and ACCESS and now I am using a R9M Lite Pro module.
The tests I´m doing right now are done with:
- R9M Lite Pro with ACCESS 1.3.0
- R9MM with ACCESS 1.3.1

kilrah, 2020-07-13:
Sure, it's just that if we attempt to reproduce we want to make sure to be on a "knoen bad" combination.

Also can you clarify that you did see the issue with the internal module too? What FWs/Rx on that?

Enurian, 2020-07-13:
When using the internal module, it was paired with an R-XSR on ACCST. Firmwares it occurred on: 2.1.0 and 2.0.1, from the R-XSR I also know 191128 (prior 2.0) and 190311. 190311 was used with OpenTX 2.2.4 (no problem) and OpenTX 2.3.2 (issue started). All firmwares were FPORT. However, I cannot not test that (easily) anymore because all quads got switched over to R9.
I reconstructed everything by looking at the logs from the time, the issue is not difficult to spot.

kilrah, 2020-07-13:
OK. 
Don't have access to an FC right now but should in a couple of days.

Enurian, 2020-07-13:
I can test only with R9 anymore, but still a little progress: I wanted to go back to 2.2.4 and the first step was to have it running on ACCST again. To my surprise, OpenTX 2.3.9 + R9 ACCST work fine!
[image]

I am sorry, contrary to what I thought, it looks like the firmwares are definitely an issue and less so OpenTX. Basically on D16 I can see the problem arising with OpenTX update from 2.2.4 to 2.3.9. However, I have not the debug setting set in these old logs, so it is a *little* more guessing. 2.4G ACCST is also a little slower packet wise and prone to more packet loss, so maybe I have misinterpreted that and it was just a bad day.
On R9, it ran fine with 2.2.4 + ACCST. I do not have a log with 2.3.2 and ACCESS unfortunately. But with 2.3.9 + ACCESS it is present. It may be an ACCESS thing after all, no idea if 2.4G on ACCESS is affected as well.

Next thing I´m going to test is if I can solve this for me by just putting an aux channel in the first slot, so I don´t care for slower updates.

3djc, 2020-07-13:
Well, it is interesting, also interesting is how the 5th channel will behave

Enurian, 2020-07-13:
Soooo, I got it working with OpenTX 2.3.9, the R9M Lite Pro on ACCESS 1.3.0 and the R9 MM on ACCESS 1.3.1.
Channel 5 had a complicated mix, so I did a quick swap with channel 6. Some aux sitting in the first channel, resulting in the map 2AER1T34.

This has *all* important channels working, here is throttle on channel 6:
[image]

Please FrSky, this is why people are having trust issues with you... if you need to gimp a channel, take the last one or something. But it´s probably just a bug and no one noticed.

Enurian, 2020-07-14:
Okay, so do you guys think this could have anything to do with OpenTX or should I close the issue?
I have mailed FrSky through their website contact form, not sure if that´s the best way to let them know about the bug (and request for a fix).

3djc, 2020-07-15:
Well, we looked through the code,and could not find anything that could cause a different sampling on the first channel, and you confirmed that with ACCST working.

We thought about a timing issue with module, but since module is not synced in R9, that would not produce a result like that.

Then that leaves only a possible bug in the  pxx2 driver on our side. At this stage, while we cannot think of what  could provoque something like that in our code;, we don't have hard proven it is not us, and proving that will no be very simple. The simplest we could do is try to find the oldest access drivers for this module and see if issue exist with those too

kilrah, 2020-07-15:
I wanted to try and have a look myself but it seems the blackbox is defective on my FC :(

closed 2021-06-04 by stale[bot]