opentx/opentx #7817: Mixing Weights with Multiplex=Add/Multiply

BeckyYork, 2020-07-14:
I'm trying to apply rates to a mix based on an input which is on a scale of min:+50 to max:+100 which is sourced from a switch.

Input Setup
Name: Rate
Source: Switch: SC
Weight: 25%
Offset: 75%

Mix 1: Source: Ail Weight:100%
Mix 1.a: Source: Rate Weight:100% Multiplex: Multiply
Mix 2: Source Ele Weight:100%
Mix 2.b: Source Rate Weight:100% Multiplex: Multiply

This works perfectly - such that when SC is in the up position Cyclic is fully up and left the Ail/Ele positions are sent as 50% instead of 100%

The problem that I'm having is that I want to apply half the rate amount to rudder and collective.
Such that when SC is in the up position Rudder and Collective are fully up and left the positions are sent as 75% instead of 100%

Ideally I'd like to do the following
Mix 1: Source: Ail Weight:100%
Mix 1.a: Source: Rate Weight:50% Multiplex: Multiply
Mix 2: Source Ele Weight:100%
Mix 2.b: Source Rate Weight:50% Multiplex: Multiply

So that half of the Rate gets applied to the Multiply rather than the full amount.
But what's happening is that it's applying the full rate first, and then dividing the whole output channel by half.

I've worked around this by setting up an additional input and applying that to the rudder and collective.
Name: RatP
Source: Switch: SC
Weight: 15%
Offset: 85%

Please change the "Weight" of sub-mixes either Add, Multiply such that only N% of the mix gets applied. Replace is fine as-is.
EG:
Input A - +100
Input B - +50
Mix A+B: Add: Weight: 100% => Output: 150
Mix A+B: Add: Weight: 50% => Output: 125
Mix A+B: Add: Weight: 25% => Output: 112.5
Mix A+B: Add: Weight: 0% => Output: 100
Mix A*B: Multiply: Weight: 100% => Output: 50
Mix A*B: Multiply: Weight: 50% => Output: 75
Mix A*B: Multiply: Weight: 25% => Output: 87.5
Mix A*B: Multiply: Weight: 0% => Output: 100
Mix ArB: Replace: Weight: 100% => Output: 50
Mix ArB: Replace: Weight: 50% => Output: 25
Mix ArB: Replace: Weight: 25% => Output: 12.5
Mix ArB: Replace: Weight: 0% => Output: 0

kilrah, 2020-07-14:
Multiply multiplies the result of everything above it. 
If you need to separate things you can use unused channels as temporary values.
You should however do your rate management in inputs.

closed 2020-07-14 by kilrah

BeckyYork, 2020-07-14:
The reason I change the rates in the mix is for the same reason people use mix for flaperons because it has SlowIn/SlowOut.

This way when the pitch rates changes from 100% to 75% my heli doesn't fall out of the sky; it gets smoothed in so I have time to adjust the input.

But I can continue with the work around and have seperate Rate Inputs for Collective, Pitch and Rudder as above.

kilrah, 2020-07-14:
Flight modes are typically used for smooth transitions.