opentx/opentx #7819: Radiomaster TX16S flashing

GotskGK, 2020-07-14:
Hi,

I've got a Radiomaster TX16S running opentx 2.3.9 with the latest companion, on Mac. I've discovered a bug which I haven't seen any reports of in the issues section. If I want to flash a receiver through the module bay, the model I have selected has to have the external module set to ON (eg r9m). Otherwise, the tx16s will not power the module bay. This means that if I had a model that used the Multi module with frsky and I wanted to flash my frsky receiver, I would have to go to setup, switch on external module in order to actually power the receiver before flashing, flash, and then switch off external module again. I suppose an easy fix would just be to switch on the module bay right before flashing? Thanks!

3djc, 2020-07-14:
Well, I'm sorry to say that, but it is not the case. I is even less the case because the power actually needs to come at the right timing for the receiver to start in bootloader (and yes I also just tried it with a blank model)

closed 2020-07-14 by GotskGK

3djc, 2020-07-14:
Why are you closing ? I can't reproduce with the elements you gaven it does not mean that there is not something wrong ?