opentx/opentx #7821: 2.1 firmware download: build specifications not working properly

jbussmann, 2020-07-15:
After years I finally found some time for the hobby and dug out my trusty Turnigy 9X, with FrSky mod and OpenTX v2.1.7 installed. Because I wanted to change the language of the firmware back to English, I tried to update to the latest and only downloadable version which should be v2.1.9.

- Companion is on v2.1.9 and I assume to download the firmware of the same version. When flashing, the downloaded image reports as v2.1.10 and refuses to flash due to hardware incompatibility.
- Flashing manually with avrdude works fine. However, the radio reports EEPROM error which is expected due to different EEPROM format. Initializing EEPROM and writing the settings with Companion in the new format from a backup works flawlessly.
- Only problem remaining: The two switches affected by the FrSky mod (AIL and THR) don't work anymore, no matter whether I flash frsky or non-frsky. This leads me to believe that my selected build options are not applied correctly. The files differ in checksum though, between selecting and un-selecting the option, so at least something must be different.

So: Is there a way I can retrieve an appropriate firmware of the abandoned 2.1 branch without going through [this build procedure](https://github.com/opentx/opentx/wiki/OpenTX-Windows-Build-Instructions)? Why is the downloaded image reporting 2.1.10? Are the selected build specifications visible somewhere in the image (I can see the version and EEPROM format in the binary file, possibly due to the stings in the menu screen). 

General information:
 - OpenTX version 2.1.9
 - Turnigy 9X with ATmega128, FrSky DHT and FrSky mod
 - OpenTX Companion version 2.1.9 on Windows 10

kilrah, 2020-07-15:
We had to hastily recreate a new build server some time ago and something likely broke and doesn't take the options into account anymore, already had a report about that

jbussmann, 2020-07-15:
Are you referring to #7230? I thought it might be related but since it downloaded something, I wasn't sure. Any chance someone might fix it or am I out of luck?

kilrah, 2020-07-15:
No, nothing to do with it. It might get fixed, but no promises.

jbussmann, 2020-08-07:
Took me a while to test things out...

**Using v2.0**
Thought about this as well but was too lazy to check all the changelogs to make sure that indeed nothing important has changed. Tried it none the less, unfortunately without success. No matter whether I download and flash with activated frsky build option or not, throttle and aileron switches still don't work. Kind of expected this...
Funny thing to note: as when downloading v2.1, the download dialog reports v2.0.9 and names the file accordingly, while the flash dialog as well as the version page on the radio report v2.0.10.

**Building it myself**
Since I'm going to do a clean install soon anyways, I thought I might give this a try. So far I haven't had much luck.
- When trying to compile for Taranis I get errors about wrong parameter format and a failed recipe for target 'sizebefore' in line 1703 of the makefile. 
- When trying to compile for 9X I get syntax errors, missing "sh" commands, wrong parameter format, a missing file and a failed recipe for target 'gccversion' in line 1710 of the makefile.

I expect most of the errors coming from missing software pieces or incompatible versions thereof. Can anyone tell me the minimal software requirement for building the v2.1 firmware only on Windows? Or am I better off installing Linux (been thinking about dual boot)?

**Sticking with OpenTX v2.1.9**
Actually I got used to the useless language on the radio because many things are a pain to configure anyways. I would be happy if I could just do all the configuration in companion. Unfortunately this doesn't work most of the time since companion writes the new format to EEPROM which is useless for the old firmware. However I had occasions where reading the configs, modifying them in companion and writing them back to the transmitter magically just worked. However I'm still working on reliably reproducing this behavior...

pki791, 2021-04-11:
Hi, is there a way to manually download the old builds from 2.1 branch? The 2.1.10 is broken.

closed 2022-04-16 by stale[bot]