opentx/opentx #7825: Minor UI issues on X10S Express

Alex00101, 2020-07-17:
1) OpenTX2.3.9 on X10S Express - Registration ID is not displayed on Model Setup screen
Steps to reproduce the behavior:
1. Select the model (set for  ISRM ACCESS) and go to Model Set Up screen
2. Scroll down to the Internal RF section.
3. Reg.ID field is not visible
4. Click 'Register' - Reg.ID is visible in the registration dialog but even after registration not shown on the Model Set Up screen.
**Expected behavior**
This may be by design, however on my X10S radio upgraded with ISRM Upgrade module (also OpenTX2.3.9) Reg.ID is always on even if model is set to ISRM D16.  Perhaps the expected behavior is to not show it for D16 and always show it for ACCESS?

2) OpenTX2.3.9 on X10S Express - "Select RX..." appears on Model Selection Dialog.
Steps to reproduce the behavior:
1. Select the model (set for  ISRM ACCESS) and go to Model Set Up screen
2. Scroll down to the Internal RF section and perform Register/Bind sequence for ACCESS.
3. Exit Model Set Up screen and long press right button for model selection dialog. 
4. "Select RX..." appears on Model Selection Dialog and goes away after radio restart.
**Expected behavior**
I think this is an unwanted remnant of the ACCESS Bind dialog.

3) OpenTX2.3.9 on X10S Express -  RX ID check behavior inconsistent between ACCESS and D16
Steps to reproduce the behavior:
1. Select the model (set for  ISRM ACCESS) and go to Model Set Up screen
2. Scroll down to the Internal RF section and try to set Receiver No to the value used in other model:
2a. If the model is set to ACCESS changing value in Receveier No field does not produce warning, however that warning i sproduced if I select Mode ISRM field end press right button 2 times.
2b. If the model is set to ACCESS changing value in Receveier No field produces the warning (as expected). The same seems to be the case on X10S radio upgraded with ISRM Upgrade module (also OpenTX2.3.9). 
Again not sure if this is by design.

[image]
[image]

3djc, 2020-07-17:
Could you avoid opening an issue with two issues inside ? That completely breaks github autoclosing feature ...

Alex00101, 2020-07-17:
Apologies.  Would you like me to go back and try to split it?

Sorry

3djc, 2020-07-17:
There are even 3 apparently, soo confusing. Yes please open one per issue

Alex00101, 2020-07-17:
Ok, doing now...

3djc, 2020-07-17:
The registration ID is shown only of different from Ownerid, and I believe behaves the same way in both radio. What could be improved is to not display it for D16

closed 2020-07-17 by 3djc