opentx/opentx #7827: X10S Express - RX ID check behavior inconsistent between ACCESS and D16

Alex00101, 2020-07-17:
OpenTX2.3.9 on X10S Express.
**Steps to reproduce the behavior:**
1. Select the model (set for  ISRM ACCESS) and go to Model Set Up screen
2. Scroll down to the Internal RF section and try to set Receiver No to the value used in other model:
2a. If the model is set to ACCESS changing value in Receveier No field does not produce warning, however the warning is produced if I select Mode ISRM field end press right button 2 times.
2b. If the model is set to D16 changing value in Receveier No field produces the warning (as expected). The same seems to be the case on X10S radio upgraded with ISRM Upgrade module (also OpenTX2.3.9). 

**Expected behavior**
I think it should be as described in 2b. (D16)

kilrah, 2020-07-17:
Do you actually have another model using the same receiver no and using ACCESS?

The system is supposed to allow duplicate numbers if the protocol is different since that naturally prevents a conflict on its own.

Alex00101, 2020-07-17:
Thanks, I will retest. Guess I was confused as it looks like the check is
preformed on different fields, depending on the protocol.

closed 2021-06-04 by stale[bot]