opentx/opentx #7836: 'Value' Widget won't change colour Radiomaster TX16s

GotskGK, 2020-07-21:
Hi,

Setup: Radiomaster TX16S, Opentx 2.3.9

When setting up widgets, in the widget settings, regardless of what colour I set it seems that the text colour for the 'value' widget is always dark blue?

3djc, 2020-07-21:
Please this place is for bug tracking. For usage question, please go to community places like rcgroups or openrx forums

closed 2020-07-21 by 3djc

GotskGK, 2020-07-21:
Surely if it won't change colour that is a bug? I mean I can adjust the R, G and B of the colour I want for the value widget but it doesn't actually carry this out. Maybe I'm wrong.

3djc, 2020-07-21:
[image]

[image]

GotskGK, 2020-07-21:
[Archive.zip](https://github.com/opentx/opentx/files/4954824/Archive.zip)
Hmmmm that's weird... this is what I get.

GotskGK, 2020-07-21:
To confirm, the same also seems to happen when using the dark theme.

kilrah, 2020-07-21:
The value always shows in blue if no data is received, when there is your color is used

loetefix, 2020-08-03:
That the value shows allways in blue if no data is still a bug because it should appear in the system "ALARM_COLOR_INDEX".
In default theme it should appear in RED and in the darkblue theme in a grey.

in the file opentx/radio/src/gui/480x272/widgets.cpp there are two lines with:

      lcdSetColor(ALARM_COLOR_INDEX);

i changed to:
    
     lcdSetColor(lcdColorTable[ALARM_COLOR_INDEX]);

and now the value appears in the right color if no data isreceived