opentx/opentx #7837: Inverse hardware

frizzle101101, 2020-07-22:
Hi so i just updated some gimbals in my opentx transmitter and did the calibration but cant see any way to inverse the hardware, i culd go to the mixxer on each model but that would be super annoying to have to inverse all my controls on all my models, im hoping theres a way to inverse the hardware so mixers dont need to be changed

3djc, 2020-07-22:
Unless you fork and modify the code accordingly, you cannot do that I'm affraid. The only supported inversion is installing Hall gimbal in an X9E

frizzle101101, 2020-07-22:
Thats silly why isnt that a feature request them?

3djc, 2020-07-22:
If it makes you feel better reopening it, but since that cannot be done at least before next major release it won’t help you

frizzle101101, 2020-07-22:
But it culd be in a maintenance release of the next major update...i will
sill do the modification and make a pull request i guess then

frizzle101101, 2021-01-19:
Havent had a chance to do a big implementation on this so i was wondering how i could enable the "horussticks" options on somthing like the t16 or any other target other than x9e

closed 2022-04-17 by stale[bot]